zmkfirmware/zmk #1766: Cannot roll PLUS EQUAL

pujvz, 2023-04-20:
I am unable to roll `PLUS` and `EQUAL`. 
- Expected output: `+=`
- Actual output: `+`

In order to achieve the desired output `+=`, `PLUS` must be released prior to pressing `EQUAL`. The same is true in reverse (`=+`).

This behavior also occurs when I try to roll `MINUS` and `UNDER`. I am aware that `KP_PLUS`, `KP_EQUAL`, `KP_MINUS` are a workaround, but this is not an option for me because the software I use for work incorrectly interprets these codes.

caksoylar, 2023-04-20:
This looks like a duplicate of #1076.

pujvz, 2023-04-20:
Thanks @caksoylar . Yes, this is a duplicate of #1076 . I did not find this issue when searching for the original issue.

closed 2023-04-20 by pujvz