zmkfirmware/zmk #1771: Feature Request: Layer Lock Key

kvietcong, 2023-04-21:
Hi, I'm coming from QMK and there was a userspace library that was incredibly helpful: https://getreuer.info/posts/keyboards/layer-lock/index.html. It allows you to lock a layer when you are accessing it through some temporary means. So if you had a momentary layer key, holding that down and then pressing this layer lock key would essentially toggle it on. The blog post I linked goes into more depth and has the implementation details. C code here: https://github.com/getreuer/qmk-keymap/blob/main/features/layer_lock.c.

This is similar to #1299 but a &df solution (changing the default layer just to lock a layer temporarily) seems too heavy handed.

petejohanson, 2023-04-21:
Note: since QMK code is GPL, it should not be used as a reference when developing ZMK features. It is important that we honor the terms of the GPL license chosen by QMK.

See https://zmk.dev/docs/development/clean-room

kvietcong, 2023-04-22:
@petejohanson The code I linked is user code rather than QMK core code and uses a different license (Apache 2.0). Should I still remove the code link and just have the blogpost link that explains how it works at a high level?

petejohanson, 2023-04-22:
> @petejohanson The code I linked is user code rather than QMK core code and uses a different license (Apache 2.0). Should I still remove the code link and just have the blogpost link that explains how it works at a high level?

Ideally, yes, using the description of the behavior is best.

bmijanovich, 2023-08-08:
I posted an example of how to achieve this on #1299 here: https://github.com/zmkfirmware/zmk/issues/1299#issuecomment-1667045948

kvietcong, 2023-08-12:
Thanks for the suggestion @bmijanovich. Sadly, I can't seem to get it working with my setup that heavily revolves around sticky keys.

bmijanovich, 2023-08-12:
Happy to try to help troubleshoot if you want.

kvietcong, 2023-08-12:
I was actually able to get it working with another day of troubleshooting. Thanks for the offer! I uploaded on this branch of my config: https://github.com/kvietcong/zmk-config/blob/layer-lock/config/the_abyss.keymap

I did some things to try cutting down on the config noise and I it works for the most part. However, this method doesn't work really well with the nice!view display and does feel a little jank with the extra noise in the config. I still appreciate the workaround though!

I will still keep this issue open as I do feel this issue warrants a dedicated feature rather than a conditional layer workaround. It'd be super nice to just have a key to lock all layers currently activated instead of defining three layers for every one functional layer.

bmijanovich, 2023-08-12:
I totally agree this should be a built-in feature. I'm actually wondering if there's a way to restrict the nice!view to only look at the first several layers to resolve this in the meantime. I have a lot of layers in my keymap for this sort of thing and it would be useful to mask them out somehow for displays.