zmkfirmware/zmk #1782: Feature Request : define the behavior of each key release after a combo was activated

Runeword, 2023-04-25:
Hi,
I am using the feature _process_combo_key_release_ in my QMK keymaps,
to define the behavior of each key release after a combo was activated.
https://github.com/qmk/qmk_firmware/blob/master/docs/feature_combo.md#customizable-key-releases

But it seems that ZMK does not have this feature.

petejohanson, 2023-04-25:
What do you practically use that feature for? I want to be sure that we are solving *problems* with our code, not duplicating features from other firmwares. It could be that the problem you are solving could fit a better way into the existing ZMK code/functionality.

Looking forward to hearing more about your use case(s).

Runeword, 2023-04-26:
Sure,

I have a 3 keys combo that activate this keys on hold : mo 4, kp LCTRL, kp LSHFT.
Once the combo is active (all 3 keys pressed), I would like LCTRL or LSHFT to be disabled when a specific key of the combo (one of these 3 keys) is released.

This way I could activate a behavior with a combo and then gives the opportunity to disable parts of this behavior or introduce new behaviors just by releasing specific keys of the combo.

theol0403, 2023-06-22:
@Runeword you might be in luck! I've submitted https://github.com/zmkfirmware/zmk/pull/1809 which might satisfy your usecase.

The only difference is that my PR allows you to specify a custom behavior according to the keys that are *still being held*, not the keys being released. 

In your case, instead of saying "I want shift to disable when the shift key is let go", you say "I want control and mt to be held if they are still being held while shift is let go". 

If this doesn't work for you, I could look into adding a `<partial-release-position &kp LSFT>` feature to my PR, I think it would be possible, but not as ideal, as it would break semantics by "tapping a key on key release to trick zmk into releasing a key that is still technically being held by the combo".

Runeword, 2023-06-24:
Thanks a lot ! I will try it as soon as possible