zmkfirmware/zmk #1784: Backlight docs are out of date

ReFil, 2023-04-26:
Since the Zephyr 3.2 update the PWM driver has changed, allowing for adjustment of both polarity and frequency. 

Whereas a previous backlight node would have the structure 

```
backlight: pwmleds {
        compatible = "pwm-leds";
        label = "Backlight LEDs";
        pwm_led_0 {
            pwms = <&pwm0 45>;
            label = "Backlight LED 0";
        };
    };
```

The changes now result in it looking like this if using the legacy pinmapping api

```
backlight: pwmleds {
        compatible = "pwm-leds";
        label = "Backlight LEDs";
        pwm_led_0 {
            pwms = <&pwm0 45 10000 PWM_POLARITY_NORMAL>;
            label = "Backlight LED 0";
        };
    };
```
Where 10000 is the period of the signal
This allows for the frequency to be adjusted, solving a known issue with an audible buzz/whine with some backlighting drive circuits 

And if using pinctrl the &pwm node looks like this
```
&pwm0 {
    status = "okay";
    pinctrl-0 = <&pwm0_default>;
    pinctrl-1 = <&pwm0_sleep>;
    pinctrl-names = "default", "sleep";
};
```

The entries in the pinctrl.dtsi file look like this for the same pin 1.13
```
pwm0_default: pwm0_default {
		group1 {
			psels = <NRF_PSEL(PWM_OUT0, 0, 45)>;
		};
	};
	pwm0_sleep: pwm0_sleep {
		group1 {
			psels = <NRF_PSEL(PWM_OUT0, 0, 45)>;
			low-power-enable;
		};
	};
```

And the backlighting node looks like this
```
backlight: pwmleds {
        compatible = "pwm-leds";
        pwm_led_0 {
            pwms = <&pwm0 0 10000 PWM_POLARITY_NORMAL>;
        };
    };
```

The labels in the backlighting are seemingly superfluous and can be removed without consequence which gets rid of some deprecation warnings.

caksoylar, 2023-06-11:
Do you think you could open a PR to fix the documentation @ReFil? Your description is pretty good but I am not very familiar with the feature to edit it myself.

ReFil, 2023-06-11:
I can give it a go but I don't really know what I'm doing with the documentation

caksoylar, 2023-06-11:
That would be great, I could PR any fixes to your branch if you don't have the [docs dev environment](https://zmk.dev/docs/development/documentation) set up.

bgkendall, 2023-08-24:
I spent all of yesterday evening trying to get backlight working on my board and feel it is important that the docs mention that the following setting will be needed for boards that support external power switching:
```
CONFIG_ZMK_RGB_UNDERGLOW_EXT_POWER=n
```
Without this the backlights will only come on when the RGB lighting is also on!

Also, this: `psels = <NRF_PSEL(PWM_OUT0, 0, 45)>;` is probably more clearly written as: `psels = <NRF_PSEL(PWM_OUT0, 1, 13)>;`. Its functionally the same, but Pin 1.13 more obviously maps to “1,13” rather that “0,45”.

ReFil, 2023-08-24:
> I spent all of yesterday evening trying to get backlight working on my board and feel it is important that the docs mention that the following setting will be needed for boards that support external [...]

What board is this? As far as I know that's not a standard behaviour and seems like a board specific quirk

> Also, this: `psels = <NRF_PSEL(PWM_OUT0, 0, 45)>;` is probably more clearly written as: `psels = <NRF_PSEL(PWM_OUT0, 1, 13)>;`. Its functionally the same, but Pin 1.13 more obviously maps to “1,13” rather that “0,45”.

This has already been updated

bgkendall, 2023-08-24:
> What board is this? As far as I know that's not a standard behaviour and seems like a board specific quirk

nice!nano in a Gherkin — the RAW and VCC pins are shorted on the board so you can only have one connected. I connected VCC, which means it is switched. I guess I would change “the following setting will be needed” to ”the following setting *may* be needed”

bgkendall, 2023-08-24:
Or, maybe: ”The following setting will be needed if your shield uses the VCC pin to power backlight LEDs:”

caksoylar, 2023-08-24:
That should probably be a warning in the underglow page, since it affects other peripherals like displays as well. Or maybe a better solution is making it conditional on the status of other features rather than defaulting it to true unconditionally. It is certainly a common enough issue.

caksoylar, 2023-12-11:
Fixed by #1894, thanks!

closed 2023-12-11 by caksoylar