zmkfirmware/zmk #1787: Turning NKRO on seems to cause nice! nano's to reset randomly

nerdo, 2023-04-30:
I'm not sure how to reproduce this issue, but I can explain my setup.

I'm using ZMK on my corne wireless with nice! nano controllers and was experiencing random resets every few minutes. It would not only reset, but a keypress would seem to get stuck (repeating excessively) until it seemed like the controller crashed and rebooted.

For a while now I have had NKRO turned on in my configuration and I didn't realize that it was the culprit until I tried to troubleshoot this resetting issue, but once I turned it off and made no other changes, the random resetting stopped. It used to reset every few minutes, but it has been several days without incident now.

My full zmk config is here https://github.com/nerdo/zmk-config-corne/

If there's some conflicting configuration in my setup, please let me know, as I would love to re-enable nkro, but it's unusable right now.

aumuell, 2023-04-30:
Could it be that the size of the `keys` array in the `zmk_hid_keyboard_report_body` struct in app/include/zmk/hid.h should be `ZMK_HID_KEYBOARD_NKRO_MAX_USAGE / 8 + 1` instead of `(ZMK_HID_KEYBOARD_NKRO_MAX_USAGE + 1) / 8`?  It seems that e.g. `select_keyboard_usage` in app/src/hid.c has the potential to write beyond this limit.

Katona, 2023-05-04:
I also experience this without NKRO being enabled. I am also using nice!nano, but no clue what can cause this.

jhorology, 2023-05-05:
Even with HKRO, I'm in same situation with BT60. I think maybe recent windows update affects this issue.