zmkfirmware/zmk #1799: Sticky layer remains active for to long if sticky key (LSHIFT) was pressed before.

hpgisler, 2023-05-12:
### Obersvation:

When press-/releasing each:
1. sticky key `LSHIFT` , then
2. sticky layer `ALA1`, then, let's say 
3. `Z`, then - within 1 second of 1. keypress - , e.g.
4. `A` then `SEMI` is sent to PC, instead of A.

### Expected Behavior:

In step 3. after `Z` has been pressed and released, further keypresses should be related to base map again, regardless whether 1s timeout since step 1. has passed or not.

Or do I miss something here?

Thanks.

### My Setup:

```
behaviors {
    skq: sticky_key_quick_release {
      compatible = "zmk,behavior-sticky-key";
      label = "STICKY_KEY_QUICK_RELEASE";
      #binding-cells = <1>;
      bindings = <&kp>;
      release-after-ms = <1000>;
      quick-release;
      ignore-modifiers;
    };
};    
```
and

```
keymap {
    compatible = "zmk,keymap";
    
    ala0 {
      label = "ALA0";
      bindings = <
      &kp N               &kp H               &kp D               &kp U               &kp O               &kp A
      &skq LSHIFT         &kp R               &kp S               &kp T               &kp SPACE           &kp E               &kp I               &mt LCTRL ESC
      &sl ALA2            &sl NAS0            &sl FUN0            &sl ALA1    
      >;
    };

    ala1 {
      label = "ALA1";
      bindings = <
      &kp J               &kp K               &kp V               &slsbsls            &kp Q               &kp SEMI  
      &trans              &kp W               &kp Y               &kp P               &kp C               &kp COMMA           &kp Z               &trans
      &sl ALA3            &sl NAS1            &trans              &trans
      >;
    };
...
};
```

theol0403, 2023-06-22:
Can you give https://github.com/zmkfirmware/zmk/pull/1812 a try? It is a little hard to visualise the problem, but it sounds like the sticky layer doesn't process the release behaviour properly, maybe something to do with quick-release. 

My PR modified the logic a little bit and might solve this?

nguyendown, 2023-07-02:
I assume you are not using ZMK main. #1745 fixed this already.