zmkfirmware/zmk #1806: Alt-Tab again

kaievns, 2023-05-19:
Hey folks, I think I'm on my single button alt-tab BS again. And I'm trying to take a different route to it this time around, and I need a bit of an advice.

I'm using a tiny 34 keys sweep keyboard, one half looks like so:

```
X X X X X
X X X X X
X X X X X
      X X
```

The bottom line has two thumb buttons, and what I'm trying to do is to simulate an alt-tab (or cmd-tab) window switching mechanics. The thumb holds down the right thumb button, that switches to another layer with a layer-tap behaviour, and that layer has a fake alt-tab button under the pinkie, like so:

```
 +-- tap this (&altab macro)
 v
AT x x x x
 x x x x x
 x x x x x
       x LT <- hold this (&lt EDIT SPACE)
```

I can define that one button alt-tab as a macro, it holds LALT down, taps TAB and then switches to another layer that basically swaps that button to a regular tab, so I could keep tapping it and list through the windows. The macro looks likes so (thanks to @danielo515 for the idea!):

```
{
  macros {
    altab: alt_tab_enter {
            label = "alt_tab_enter";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            wait-ms = <10>;
            tap-ms = <20>;
            bindings =
                <&macro_press &kp LALT>,
                <&macro_tap &kp TAB &to ALTAB>;
     };
  };
};
```

It all works pretty great, except one thing. When I release that thumb button the `LALT` doesn't get released to finish the alt-tab-tab-tab sequence and the whole thing is kind of stuck in there.

I could define another special macro to release it like Daniel did and assign it to other buttons, but it kind of breaks the feeling of natural alt-tab movement. And what I would like to do is release that `LALT` when the thumb `&lt` is released.

Any thoughts how I could do that? Is there any ways to combine `&lt` with a macro or something maybe?

Thank you!

caksoylar, 2023-05-19:
Related: #997. 

There are a couple solutions involving PRs that I am aware of, I haven't myself figured out a good solution for stock (and I don't know if I would prefer it if it involved an extra layer definition:
- Using the swapper PR #1366: https://github.com/zmkfirmware/zmk/issues/997#issuecomment-1314549032
- Using the new `macro_pause_for_layer` modifier for macros: #1804

kaievns, 2023-05-20:
Thanks @caksoylar I'm following your PR, been waiting for it for a while actually ;)

I'm just trying to find another path to it.

Is there a way to maybe send a macro into a hold-tap thing? like `&ht &macro SPACE` and then use the macro to jump layers and wait for the release?

caksoylar, 2023-05-20:
You could assign a macro to the layer change key (`LT` in your example) that would release the alt key on release (regardless of whether a alt press was sent earlier) but you would also need it to turn off the special alt-tab layer, right?

Right now there is no behavior to turn off a layer directly, only to toggle it. You could instead use the `&to` behavior to a base layer, which would turn off all other layers. This won't be ideal if you have multiple base layers that you can reach this from, like separate QWERTY/Colemak ones.

Given that, the hold-tap key `LT` that activates the layer that contains `AT` could look like the following with the corresponding macro definition (not tested):
```dts
    // macro to activate layer 1, then release alt and move to layer 0 on key release
    mo_at: momentary_release_alt {
        compatible = "zmk,behavior-macro";
        label = "momentary_release_alt";
        #binding-cells = <0>;
        wait-ms = <0>;
        tap-ms = <0>;
        bindings = <&macro_press &mo 1>
                 , <&macro_pause_for_release>
                 , <&macro_release &kp LALT>
                 , <&macro_tap &to 0>;
    };

    // use below with a dummy value for first parameter: &lta 0 RET
    lta: layer_tap_release_alttab {
        compatible = "zmk,behavior-hold-tap";
        label = "layer_tap_release_alttab";
        #binding-cells = <2>;
        flavor = "tap-preferred";
        tapping-term-ms = <200>;
        bindings = <&mo_at>, <&kp>;
    };

```

kaievns, 2023-06-19:
Apologies for a lag with the response here, Cem. Thanks for the snippets, it worked like a charm!

I only have to layouts on the keyboard, so I have copied the snippets twice and it all worked out fine. Could probably tidy it up a bit with a code generator, but it's fine as it is.

Posting a link here to my version for posterity https://github.com/kaievns/zmk-config/blob/944ecf502cc73f186c83b6228f1225cf6ece5b61/config/cradio.keymap

closed 2023-06-19 by kaievns