zmkfirmware/zmk #1831: MT and tap dance delay

nosleduc, 2023-06-11:
Hi

I'm using the glove80. I noticed a delay when combining mt and tap dance.
1) 
On my j key, I have defined a tap dance with the following behaviour:
Hold => LGUI
Tap => j
Double tap = >  macro   => ( )

2)
On my left space bar, I have defined a mt with space for tap and shift for hold

When I'm holding the space bar and press j, I get most of the time j instead of J. To get J, I need to press longer the space bar.
If I'm doing the same thing but with a key without a tap dance, I get the shifted version right away.

I might be wrong but it seems for me that the firmware may use the tapping term of the tap dance instead of of the one defined by the mt.  

Has anyone experienced the same issue ?

caksoylar, 2023-06-11:
This behavior sounds expected to me. `&mt` is listening for an interrupting key press to make a hold vs. tap decision (with `balanced` and `hold-preferred` flavors), but tap dance doesn't resolve to `j` until its tapping term expires, so if you release `&mt` before that happens it doesn't observe the interrupting `j`.

Not sure how this would be improved easily without very closely coupling hold-tap and tap dance decision logic.

nosleduc, 2023-06-14:
Thanks for answering me.

Setting tap dance on h,j, and k is very convenient for me. 
As a developer, I use a lot (,{,[.

For example, I have configured my K key as
Tap => k
Double tap = >  macro => { followed by a Enter
Hold => Left control.

I would like to also set some home row on the left part of the keyboard. But with the "issue" I mentionned above,  holding the right space to get the shift modifier is not practical in fact when home row are set. You need to slow down to be sure that you will get the shifted version

Is there a workaround ?

caksoylar, 2023-06-14:
I am not aware of a workaround for this particular issue. If you ask my advice I would recommend a different approach for symbols in general, such as placing them on the home row on a layer, or on combos ([which is what I do](https://github.com/caksoylar/zmk-config/)).

In general I'd avoid tap dances on alpha keys because their bigrams can occur during natural typing, although in this particular case `hh`/`jj`/`kk` are not very common, at least in English.

nosleduc, 2023-06-14:
Thanks, I keep a qwerty layout and I choose these three keys because there almost never used with two consecutive occurences. It allows me keeping my thumbs at the same place and avoid to move them to switch to a different layer.
I will try with combos

closed 2023-06-14 by nosleduc