zmkfirmware/zmk #1832: USB CDC ACM "ring buffer full"

xudongzheng, 2023-06-11:
Currently ZMK can end up with a ton of `<inf> usb_cdc_acm: Ring buffer full, drain buffer` messages despite it being turned off https://github.com/zmkfirmware/zmk/blob/19d883c/app/Kconfig#L433-L435. This appears to be an issue with Zephyr 3.2 since the underlying `LOG_LEVEL` doesn't work properly https://github.com/zmkfirmware/zmk/blob/19d883c/app/Kconfig#L433-L435.

This appears fixed in Zephyr 3.4 RC2 (not sure about Zephyr 3.3). A temporary workaround is defining `CONFIG_USB_CDC_ACM_LOG_LEVEL_WRN=y`.

```
#define LOG_LEVEL 0
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(test);

void main(void) {
	LOG_INF("Hello World! %s", CONFIG_BOARD);
}
```

caksoylar, 2023-06-11:
In that case would it make sense to `select USB_CDC_ACM_LOG_LEVEL_WRN` under `ZMK_LOGGING` until the upgrade, perhaps?

xudongzheng, 2024-02-15:
Closing as this is resolved with the Zephyr 3.5 upgrade.

closed 2024-02-15 by xudongzheng