zmkfirmware/zmk #1842: When using SETTINGS_NVS node is unable to re-pair with PC

DuMaM, 2023-06-19:
Hi,

I'm facing this issue for quite a while. I'm unable to pair with my keyboard after unpairing it.
Furthermore, I tested this with a few different boards:
* weact_esp32c3
* particle_xenon

No matter how I try to do this, it does not work, until I clean up the whole flash or remove this setting from the flash.

## Preconditions:
* Working CONFIG_SETTINGS_NVS
* Split keyboard

## Reproducing scenario:
a) pair with PC using central node
b) unpair it in PC after a while
c) try to pair again

## Result:
The device is refusing to pair again and there are some errors in windows.
On the device, there is nothing suspicious in the logs.

## Expected result
The board should accept new pairing from a new device which is already paired.
Thats how my all head phones works.

## Notes:
I think the reason behind it is that split keyboard may confuse a second half profile stored in settings with my PC.
According to zmk app code, we need an active profile to be set to repair it.

petejohanson, 2023-06-19:
Have you read through https://zmk.dev/docs/features/bluetooth#profiles ?

Are you clearing the profile on the keyboard side after unpairing from Windows?

petejohanson, 2023-06-19:
> Thanks for fast reply.
> Yes, I read this doc.  For me, it's a bit unnatural because most of Bluetooth devices do not work that way. They rather use a FIFO approach. 

They do. There are security reasons not to do this, but I am right in the middle of some work to support this style of interaction.

DuMaM, 2023-07-01:
Thats great news

closed 2023-07-01 by DuMaM