zmkfirmware/zmk #1843: XIAO BLE doesn't turn off Bluetooth when entering Deep Sleep

UnchartedBull, 2023-06-19:
I have a little 6 key macro pad running ZMK. It's based on the XIAO nrf52840 (the non-sense variant). I'm using direct wiring for the switches.

Everything works now, the only Problem is, that the controller doesn't turn off Bluetooth when it enters Deep Sleep. I set the deep sleep timout down to a minute and verified with USB Logging that the board is actually entering Deep Sleep as indicated by this line being printed repeatedly:

```
zmk_usb_get_conn_state: state: 3
```

I have absolutely no idea why it doesn't disable Bluetooth then though. The mac still shows it as connected and keypresses get recognised immediately (so not the typical wakeup thing). Happens on two different controllers.

Here is my config if that helps: https://github.com/UnchartedBull/zmk-config-debugpad

Thanks!

petejohanson, 2023-06-19:
Enabling USB logging may also prevent deep sleep. Have you tried with that off?

UnchartedBull, 2023-06-19:
Yes USB logging is off now. I've also tried disabling DCDC mode, just in case this was due to a hardware fault in the DCDC circuit causing a reset, but all to no avail.

Two more things come to mind now though:
- I'm not using any external power config (since there are no smart LEDs or similar connected), does this maybe mess around with the DEEP_SLEEP?
- You've mentioned that ZMK does not enter Deep Sleep if USB is connected. For the XIAO everything is driven through a 3.3V voltage regulator, so the detection if USB is connected might be confused? Does the detection somehow check if communication over USB exists or is it checking some voltage levels?

UnchartedBull, 2023-06-20:
Ok quick update:

Setting `CONFIG_ZMK_USB` and `CONFIG_USB_DEVICE_STACK` to 'n' solved the issue. As far as I could tell from the code this effectively disables the check for an active USB connection before going into the Soft Off state. Thanks for the help and putting me on the right track with the 5V thing. As I mentioned though I have nothing connected to the 5V line, so this is probably an internal issue with the controller.

From my limited understanding of the codebase the problem probably is somewhere in this line https://github.com/zmkfirmware/zmk/blob/e686fce4d917b708da7f2baba4115043d3f0deab/app/src/activity.c#L27 for the XIAO BLE.

For me this is sufficient, since I don't need USB connectivity. It still would be nice for everything to work as intended. If you want me to debug something let me know.

UnchartedBull, 2023-06-26:
Final update, after some more investigating I found that having the switches wired to 3.3V and then do a PULL_DOWN for the GPIOs causes quite a lot of voltage on the USB pin. There should be a diode to prevent that, but oh that doesn't work. And even with BT being able to sleep with the hack above the battery lasted only 5 days in sleep (350 mAh, should last well over a year in sleep)

After rewiring the switches to GND and doing the GPIOs with PULL_UP everything just works. Sleep mode works just fine with the USB_DEVICE_STACK enabled.

So my fault, never did direct GPIO wiring (only matrix). Might be an option to add this to the documentation somewhere that one side should be GND for direct wiring.

closed 2023-06-26 by UnchartedBull