zmkfirmware/zmk #1845: Autoshift 2+ states

jkavalec, 2023-06-22:
Is there any way to compose Autoshift behavior with more than two states?

I saw some options to implement it for normal Autoshift like 'a' -> 'A', but I want something like:

< 150 ms -> 'a'
< 300 ms -> 'A'
< 500 ms -> 'Ctrl+a'
< 900 ms  -> 'Ctrl+Shift+A'

I know that I can achieve similar result with Tapdance, but I prefer holding key for duration (on QMK, which I intend to switch for ZMK)

theol0403, 2023-06-22:
Would probably require a new API for autoshift. However, would't this be slightly difficult to use, without any live feedback? I would never know how long to hold.

Actually, I have a crazy idea that might work. It involves some magic combination of macros, hold-taps, tap-dance, retro-tap, hold-while-undecided, and a ton of boilerplate and combo magic for getting around the devicetree syntax. 

idea 1 (nested hold-taps): 
- tap (hold-tap: a 150 (hold-tap: A 150 (hold-tap: C-A 200 C-S-A))

idea 2 (live feedback):
- macro:
  - press (hold-tap: a 150 A)
  - press (hold-tap: none 300 (macro: wait-for-release bspc C-a))
  - wait-for-release
  - release hold-taps 

idea 3 (auto tap-dance):
- td = (tap-dance: a A C-a C-S-A)
- macro:
 - press (hold-tap-while-undecided: none 150 (tap: td)) or (hold-tap-retro-tap: td 150 (tap: td))
 - press (hold-tap: none 300 (tap: td))
 - press (hold-tap: none 500 (tap: td))

All of these would require extensive use of macros because of the devicetree syntax.

Good luck!!

jkavalec, 2023-06-23:
Thank you for some good info, I will certainly try these before trying to implement new behavior.

I would probably go as far as combining this expanded Autoshift with tapdance.

Regarding live feedback - this may be specific to the switches in use, and may be inconvinient with clicky ones, however with Cherry Browns and few days of getting used to it, it gets into muscre memory pretty fast, it goes like this for 4 action setup across the keymap:
- I quickly tap, I get lowercase
- I press the key 'seriously' I get  uppercase
- I perss the key like I really mean it - I get key with modifiers 1 - these are generally common things like switching to named app, initiating search or often used action
- I press The key like above and deliberately keep it pressed just for an instant, this is for infrequent commands, like commit, push, settings screen etc (I dont think I initiated this kind of action by accident 

For layers of Autoshift can be pretty easily mapped on this kind of switch and after you learn it, it is very convenient to use

Hardest part is differentiate between 2 and 3 but it is a nonissue after a few days.

Once you fine tune this (everyone will have little different timing), you can get great accuracy of these actions