zmkfirmware/zmk #1847: C_BRI_UP not working

glyh, 2023-06-24:
I bind C_BRI_UP to this:
```
        bridn_briup: bridn_briup {
            compatible = "zmk,behavior-mod-morph";
            label = "bridn_briup";
            #binding-cells = <0>;
            bindings =
                <&kp C_BRI_DN>,
                <&kp C_BRI_UP>;

            mods = <(MOD_LSFT|MOD_RSFT)>;
        };
```

But it's not working

glyh, 2023-06-24:
I also have problem for C_VOL_UP, using K_VOL_UP is fine for me. BTW C_BRI_DN works fine.

caksoylar, 2023-07-01:
There can be a few different issues here. Is `&kp C_BRI_UP` working on a layer on its own?

In addition, not all "keyboard" and "consumer" keycodes listed in https://zmk.dev/docs/codes will be listened to by the operating system; we have the columns showing testing results per OS on that page. 

`CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL` and `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC` settings  in https://zmk.dev/docs/config/system#hid will also affect which consumer keycodes can be sent properly. You can experiment with setting either one to `=y` and see if it changes the result.

caksoylar, 2023-12-11:
Closing since there is no follow up.

closed 2023-12-11 by caksoylar