zmkfirmware/zmk #1848: corne: left display not working

gnapiorkowski, 2023-06-24:
Hello,
Recently I've rebased my local zmk repo to update my keyboard and it looks like left corne ( zmk/tree/main/app/boards/shields/corne ) display isn't working. tested with a fresh corne-config with just the display line uncommented: https://github.com/gnapiorkowski/zmk-config-corne/actions/runs/5366593277
I can't figure out what caused it, because there has been quite a few commits since last time (January 6) I have changed anything.

shawnbitzaro, 2023-06-30:
I'm facing this too and was about to post an issue.

caksoylar, 2023-07-01:
Since it hasn't been reported more generally, it could be a `nrfmicro_13` specific issue due to the large Zephyr 3.2 upgrade. 

@shawnbitzaro do you also use a nrfmicro? 

Does it work OK for the right half both of you? Can you make sure that external power is turned on through the [`&ext_power EP_ON`](https://zmk.dev/docs/behaviors/power) binding, by pressing it, waiting one minute for the setting to be saved to flash, then resetting the keyboard (or turning it off then on)?

lsshawn, 2023-07-13:
I'm using nrfmicro. Right side works for me.

The weird thing with me is if I set `CONFIG_ZMK_DISPLAY=y`, my left side fails to flash. I drag the files into it and it doesn't unmount.

But on the right side, it flashes ok and display shows up.

Config here: https://github.com/lsshawn/corne-zmk

gnapiorkowski, 2023-10-30:
Same here, I have two identical cornes, one with nrfmicros and one with nice!nanos. I use the same config for both. Nice!nano one works as expected, but for the nrfmicro the left side oled display doesn't work (right side is fine). Checked the display, checked the connections... Display is getting 3.3V.
Config here:  https://github.com/gnapiorkowski/zmk-config-corne

closed 2024-02-27 by caksoylar