zmkfirmware/zmk #1850: TapDance and Mt key sequence order not preserved

nosleduc, 2023-06-25:
I am trying to customize my glove80 with a tap dance and a mt behaviour but in most cases it does not work as expected.
So  I made a simple example:

TD1:TD1 {
    compatible = "zmk,behavior-tap-dance";
    label = "TD1";
    #binding-cells = <0>;
    tapping-term-ms = < TD_TAPPING >;
    bindings = <&mt LGUI A>, <&kp B>;
};
is placed on the A key

Holding A must activate LGUI, a tap must output A and two taps must output B.
If I'am tapping quickly A and S, I got "sa" instead "as". The other behaviours are correct.
In fact, for any interrupted keys, the interrupted key is typed first instead of being handled later.

caksoylar, 2023-07-01:
If you nest it the other way around then the interrupt detection will work better. This would look like a tap dance e.g. `&td` with `bindings = <&kp A>, <&kp B>;` and a custom hold-tap (e.g. `&nt`) with `bindings = <&kp>, <&d>;` used with `&ht RGUI 0` in your keymap.

nosleduc, 2023-07-04:
Thanks for your answer.
I do not see how I can keep the double tap behaviour. Could you share the code you suggest ?

caksoylar, 2023-07-04:
Here is an example:
```dts
/ {
    behaviors {
        // tap dance, A on single B on double tap
        td_ab: td_ab {
            compatible = "zmk,behavior-tap-dance";
            label = "td_ab";
            #binding-cells = <0>;
            tapping-term-ms = <200>;
            bindings = <&kp A>, <&kp B>; 
        };
        // hold-tap, key on hold, AB tap dance on tap
        ht_td: ht_td {
            compatible = "zmk,behavior-hold-tap";
            label = "ht_td";
            #binding-cells = <2>;
            flavor = "hold-preferred";
            tapping-term-ms = <200>;
            bindings = <&kp>, <&td_ab>;
        };
    };

    keymap {
        compatible = "zmk,keymap";

        default_layer {
            bindings = <
                &ht_td LGUI 0
            >;
        };
    };
}
```

nosleduc, 2023-07-21:
Thanks, it works better

closed 2023-07-21 by nosleduc