zmkfirmware/zmk #1856: build issue: __device_dts_ord_DT_CHOSEN_zephyr_display_ORD

osman, 2023-07-04:
Hi,

I need some help debugging this error. It seems to be telling me that display drivers are not loaded, but I'm not sure where to go next.

```
FAILED: CMakeFiles/app.dir/src/display/main.c.obj 
  ccache /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DLV_CONF_INCLUDE_SIMPLE=1 -DLV_CONF_PATH=../zephyr/lv_conf.h -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/__w/zmk-config/zmk-config/zmk/app/include -I/__w/zmk-config/zmk-config/zmk/zephyr/include -I/__w/zmk-config/zmk-config/zmk/app/build/zephyr/include/generated -I/__w/zmk-config/zmk-config/zmk/zephyr/soc/arm/nordic_nrf/nrf52 -I/__w/zmk-config/zmk-config/zmk/zephyr/soc/arm/nordic_nrf/common/. -I/__w/zmk-config/zmk-config/zmk/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -I/__w/zmk-config/zmk-config/zmk/zephyr/subsys/bluetooth -I/__w/zmk-config/zmk-config/zmk/zephyr/subsys/usb/device -I/__w/zmk-config/zmk-config/zmk/zephyr/subsys/settings/include -I/__w/zmk-config/zmk-config/zmk/modules/hal/cmsis/CMSIS/Core/Include -I/__w/zmk-config/zmk-config/zmk/modules/hal/nordic/nrfx -I/__w/zmk-config/zmk-config/zmk/modules/hal/nordic/nrfx/drivers/include -I/__w/zmk-config/zmk-config/zmk/modules/hal/nordic/nrfx/mdk -I/__w/zmk-config/zmk-config/zmk/zephyr/modules/hal_nordic/nrfx/. -I/__w/zmk-config/zmk-config/zmk/modules/lib/gui/lvgl/zephyr/../src -I/__w/zmk-config/zmk-config/zmk/modules/crypto/tinycrypt/lib/include -I/__w/zmk-config/zmk-config/zmk/app/drivers/sensor/battery/. -isystem /__w/zmk-config/zmk-config/zmk/zephyr/lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -fno-strict-aliasing -Os -imacros /__w/zmk-config/zmk-config/zmk/app/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/zmk-config/zmk-config/zmk/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/zmk-config/zmk-config/zmk/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/zmk-config/zmk-config/zmk=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -Wfatal-errors -MD -MT CMakeFiles/app.dir/src/display/main.c.obj -MF CMakeFiles/app.dir/src/display/main.c.obj.d -o CMakeFiles/app.dir/src/display/main.c.obj -c /__w/zmk-config/zmk-config/zmk/app/src/display/main.c
  In file included from /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/toolchain/gcc.h:88,
                   from /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/toolchain.h:50,
                   from /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/kernel_includes.h:19,
                   from /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/kernel.h:17,
                   from /__w/zmk-config/zmk-config/zmk/app/src/display/main.c:7:
  /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/device.h:104:39: error: '__device_dts_ord_DT_CHOSEN_zephyr_display_ORD' undeclared here (not in a function)
    104 | #define DEVICE_NAME_GET(name) _CONCAT(__device_, name)
        |                                       ^~~~~~~~~
  /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/toolchain/common.h:132:26: note: in definition of macro '_DO_CONCAT'
    132 | #define _DO_CONCAT(x, y) x ## y
        |                          ^
  /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/device.h:104:31: note: in expansion of macro '_CONCAT'
    104 | #define DEVICE_NAME_GET(name) _CONCAT(__device_, name)
        |                               ^~~~~~~
  /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/device.h:264:37: note: in expansion of macro 'DEVICE_NAME_GET'
    264 | #define DEVICE_DT_NAME_GET(node_id) DEVICE_NAME_GET(Z_DEVICE_DT_DEV_NAME(node_id))
        |                                     ^~~~~~~~~~~~~~~
  /__w/zmk-config/zmk-config/zmk/zephyr/include/zephyr/device.h:282:34: note: in expansion of macro 'DEVICE_DT_NAME_GET'
    282 | #define DEVICE_DT_GET(node_id) (&DEVICE_DT_NAME_GET(node_id))
        |                                  ^~~~~~~~~~~~~~~~~~
  /__w/zmk-config/zmk-config/zmk/app/src/display/main.c:24:39: note: in expansion of macro 'DEVICE_DT_GET'
     24 | static const struct device *display = DEVICE_DT_GET(DT_CHOSEN(zephyr_display));
        |                                       ^~~~~~~~~~~~~
  compilation terminated due to -Wfatal-errors.
```

I am trying to use the Xiao BLE with a Sharp Memory Display with the following overlay

```
&xiao_spi {
	status = "okay";

	cs-gpios = <&xiao_d 7 GPIO_ACTIVE_LOW>, <&xiao_d 6 GPIO_ACTIVE_HIGH>;

	shifter: 595@0 {
		compatible = "zmk,gpio-595";
		status = "okay";
		gpio-controller;
		spi-max-frequency = <200000>;
		reg = <0>;
		label = "4HC595";
		#gpio-cells = <2>;
		ngpios = <8>;
	};
	ls0xx: ls0xx@1 {
        compatible = "sharp,ls0xx";
        label = "DISPLAY";
        spi-max-frequency = <2000000>;
        reg = <1>;
        width = <144>;
        height = <168>;
        // disp-en-gpios = <&gpio0 23 GPIO_ACTIVE_HIGH>;
    };
};

/ {
	chosen {
		zmk,kscan = &kscan;
		zmk,matrix_transform = &transform;
		zmk,display = &ls0xx;
	};
}
```

petejohanson, 2023-07-04:
The chosen node comes from upstream Zephyr, so is called `zephyr,display`.

osman, 2023-07-04:
Thanks! I knew it must be a silly mistake :)

petejohanson, 2023-07-04:
Closing, feel free to re-open if that doesn't fix it.

closed 2023-07-04 by petejohanson