zmkfirmware/zmk #1860: Encoder docs are out of date

joelspadin, 2023-07-07:
https://zmk.dev/docs/features/encoders has no information about adjusting properties for different types of encoders.

https://zmk.dev/docs/development/new-shield#encoders is missing the change from `resolution` to `steps` and doesn't show anything about `triggers-per-rotation` on the sensor node.

https://zmk.dev/docs/config/encoders is also missing those changes. We should probably add the `sensors` node configuration to that page too.

caksoylar, 2023-07-09:
> https://zmk.dev/docs/features/encoders has no information about adjusting properties for different types of encoders.

Can you elaborate a bit further on what properties?

> https://zmk.dev/docs/config/encoders is also missing those changes. We should probably add the sensors node configuration to that page too.

Would it make sense to convert that page to a "Sensors" page instead that has a "Encoders" section?

joelspadin, 2023-07-09:
> Can you elaborate a bit further on what properties?

`steps` for the alps,ec11 node and `triggers-per-rotation` for the sensors node and child nodes.

> Would it make sense to convert that page to a "Sensors" page instead that has a "Encoders" section?

Likely yes, though we don't yet support any other types of sensors.

closed 2023-11-03 by caksoylar