zmkfirmware/zmk #1884: Plugging into USB borks BLE on macOS

kaievns, 2023-07-24:
Hey forks, how are you?

I've been having this weird issue for the past few months, and I thought I'd report it.

Basically, I have a split 2 nice!nano v2 keyboard. It works perfectly fine after flashing, BLE and all. _but_ the moment I plug it into my MacBook for a charge, macOS stops seeing it with bluetooth. It works fine on USB, but when I unplug it, it doesn't reconnect. MacOS just doesn't see it anymore. Deleting the device from the list of paired devices and trying to re-pair doesn't work either.

Turning the keyboard off and on again doesn't do anything.

The only way to get it back that works for me is to flash the reset firmware and refresh the shields a fresh. Well, until the next charge that is.

Weirdly enough, this doesn't affect Windows. I have the keyboard paired on a second channel to a windows machine, and that keeps working fine.

Here is my config for your perusal https://github.com/kaievns/zmk-config

if there is a quick fix for this, I would love to know about it, thanks

SethMilliken, 2023-07-24:
https://zmk.dev/docs/behaviors/outputs

kaievns, 2023-07-24:
thanks muchly!

closed 2023-07-24 by kaievns

myst729, 2023-11-21:
> https://zmk.dev/docs/behaviors/outputs

Is it possible to set bluetooth first without pressing a "&out OUT_BLE" key, something like `CONFIG_ZMK_DEFAULT_OUTPUT="ble"`?

caksoylar, 2023-11-21:
> > https://zmk.dev/docs/behaviors/outputs
> 
> Is it possible to set bluetooth first without pressing a "&out OUT_BLE" key, something like `CONFIG_ZMK_DEFAULT_OUTPUT="ble"`?

No, there is no config for that. But like the [docs mentions](https://zmk.dev/docs/behaviors/outputs#behavior-binding) your preference will be saved into flash and remembered once you select it.