zmkfirmware/zmk #1887: Turning on USB logging interferes with TX/RX pins on Xiao BLE

caksoylar, 2023-07-31:
I have this issue where turning on USB logging via `CONFIG_ZMK_USB_LOGGING=y` prevents the matrix pin I have at RX (D7/P1.12) from working. I am guessing this is because they are assigned to UART pins at https://github.com/zmkfirmware/zephyr/blob/v3.2.0%2Bzmk-fixes/boards/arm/seeeduino_xiao_ble/seeeduino_xiao_ble-pinctrl.dtsi#L7 and logging turns it on via https://github.com/zmkfirmware/zephyr/blob/v3.2.0%2Bzmk-fixes/boards/arm/seeeduino_xiao_ble/seeeduino_xiao_ble.dts#L17.

I see some shield configs deleting these properties, e.g. https://github.com/zmkfirmware/zmk/blob/main/app/boards/shields/hummingbird/hummingbird.overlay#L13. Should this be something that every shield does, or would it be better to solve it at the `seeeduino_xiao_ble.overlay` level?

caksoylar, 2023-07-31:
Actually I didn't have any luck with enabling USB logging with those nodes deleted, the stock `hummingbird` shield fails building too:
```sh
> west build -p -d build/hummingbird -b seeeduino_xiao_ble -- -DSHIELD=hummingbird -DCONFIG_ZMK_USB_LOGGING=y
[...]
/home/.../zmk/zephyr/include/zephyr/device.h:104:39: error: '__device_dts_ord_DT_CHOSEN_zephyr_console_ORD' undeclared here (not in a function)
[...]
```

xudongzheng, 2023-07-31:
I believe the issue is that USB logging enables UART, which causes this UART https://github.com/zmkfirmware/zephyr/blob/v3.2.0%2Bzmk-fixes/boards/arm/seeeduino_xiao_ble/seeeduino_xiao_ble.dts#L64 to be enabled internally within Zephyr. 

My initial guess is `UART_CONSOLE` https://github.com/zmkfirmware/zmk/blob/main/app/Kconfig#L438C12-L438C24 needs to be removed.

caksoylar, 2023-07-31:
Thanks, FWIW, removing `select UART_CONSOLE` gets the pins working again (but only when combined with deleting the `chosen` properties) and gets the `hummingbird` shield building. But I guess it is necessary for logging, since the COM port shows up but nothing gets logged in this case.

xudongzheng, 2023-07-31:
Two small observations:

- https://github.com/zmkfirmware/zmk/blob/main/app/boards/arm/nice_nano/nice_nano.dtsi#L19 may be needed for USB logging to work.
- https://github.com/zmkfirmware/zephyr/blob/v3.2.0%2Bzmk-fixes/boards/arm/seeeduino_xiao_ble/seeeduino_xiao_ble_defconfig#L19-L20 also has `CONFIG_UART_CONSOLE=y`

PR #2672 (2024-11-25): fix(boards): Disable uart serial node in Xiao BLE by default
`xiao_serial` node corresponds to `uart0`, which uses D6 and D7 pins by default on Xiaos. When logging or Studio USB snippets are used, they turn on `CONFIG_SERIAL` and this prevents those pins from being used for other things, even though both use serial over USB. 

For RP2040 this is disabled in https://github.com/zmkfirmware/zmk/blob/main/app/boards/seeeduino_xiao_rp2040.overlay#L7, for Hummingbird shield this was disabled at the shield level but was (mistakenly?) removed in https://github.com/zmkfirmware/zmk/commit/e0339a2a57c33b [... cut]
closed 2024-11-25 by petejohanson