zmkfirmware/zmk #1897: Backlight with SK6812MINI-E	doesn't work, build failed

t18n, 2023-08-10:
[My repo](https://github.com/t18n/zmk-config-corne)
[Failed workflows](https://github.com/t18n/zmk-config-corne/actions): #76, #77

I am trying to enable backlight on my **Cherry v3 Corne** but the build always break. Above is the code and my build.

Someone on Discord said "Backlighting is for single colour LEDs, you cant use SK6812/WS2812s for it", but I bought it as instructed [here](https://github.com/foostan/crkbd/blob/main/corne-cherry/doc/v3/buildguide_en.md)

How can I configure to make that works?

SethMilliken, 2023-08-10:
Eliminate all of your backlight `.conf` settings. Get rid of the nice!nano overlay. Add a stanza to your keymap like this:
```
&led_strip {
    chain-length = <21>;
};
```
where the number is your total count of LEDs on each half.

t18n, 2023-08-10:
@SethMilliken I have done that but the backlight is still not on :(

t18n, 2023-08-10:
I found out that it was partly soldering problem. After fixing the solder joins and adding the code, all light is now working 🥳 Thank you so much @SethMilliken.

By the way, is there a way to separate the configuration for the backlight and the glow light? If not, can I configure individual pin so that I have each led display the color I wanted?

closed 2023-08-14 by caksoylar