zmkfirmware/zmk #1900: Bluetooth issue with Intel Mac

johnbailon, 2023-08-14:
Weird issue with my Corne nice nano v2 build. using latest firmware, can't seem to connect via bluetooth to my Intel Mac running Ventura (M1 Ventura and iPhone connect fine) but when I flash a firmware built in 2022 it connects properly. This issue has been discussed on my discord [thread](https://discord.com/channels/719497620560543766/1140492596838420480) and apparently another user reported it on discord on a separate [thread](https://discord.com/channels/719497620560543766/719909884769992755/1131638872212766790) where discord user _laudans_ were able to pin point the commit where this bug surfaced. Reporting here to notify developers of this issue.

infernalmaster, 2023-08-22:
this is old known issue
https://github.com/zmkfirmware/zmk/issues/278

stefansturlu, 2024-02-12:
I had this problem as well after the Zephyr 3.2 update, where I couldn't connect to my old Intel MacBook but was able to connect to a M1 Mac. 
But after the Zephyr 3.5 update that was merged last week, everything started working properly again!
So this issue can probably be closed

johnbailon, 2024-02-15:
Can confirm that this is fixed with the [latest update](https://zmk.dev/blog/2024/02/09/zephyr-3-5). Closing.

closed 2024-02-15 by johnbailon