zmkfirmware/zmk #1911: Global quick tap does not prevent accidental triggering of hold behavior in some cases

danmou, 2023-09-01:
I use home row mods and to prevent accidental triggering during fast typing I've enabled the global-quick-tap feature. This does work in some cases, but not always. The problem appears to happen when first a HRM key is pressed down and then another key is pressed **and** released **before** the HRM key is released. The scenario where this tends to happen is usually if I intend to tap the HRM-enabled key (e.g. S) and but accidentally bump an adjacent key (e.g. W) as well. This often results in Ctrl+W (closing my tab) instead of just an extra character which I can delete. I've recorded some log snippets which demonstrate the issue. I'm on latest master (commit 185457bc).
 
<details><summary>Config</summary>

    behaviors {
        bhm: balanced_homerow_mods {
            compatible = "zmk,behavior-hold-tap";
            label = "HOMEROW_MODS";
            #binding-cells = <2>;
            tapping-term-ms = <200>;
            quick-tap-ms = <125>;
            flavor = "balanced";
            bindings =
                <&kp>,
                <&kp>;

            global-quick-tap;
        };
    };

</details> 
 
<details><summary>Press S, press Q, release Q, release S: Ctrl+Q (undesired)</summary>

[00:12:59.687,133] <dbg> zmk: kscan_matrix_read: Sending event at 1,2 state on
[00:12:59.687,255] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 2, position: 14, pressed: true
[00:12:59.687,286] <dbg> zmk: position_state_changed_listener: 14 bubble (no undecided hold_tap active)
[00:12:59.687,316] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 14, binding name: HOMEROW_MODS
[00:12:59.687,377] <dbg> zmk: on_hold_tap_binding_pressed: 14 new undecided hold_tap
[00:12:59.688,140] <dbg> zmk: kscan_matrix_read: Sending event at 0,1 state on
[00:12:59.688,232] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 1, position: 1, pressed: true
[00:12:59.688,293] <dbg> zmk: position_state_changed_listener: 14 capturing 1 down event
[00:12:59.688,323] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:12:59.781,127] <dbg> zmk: kscan_matrix_read: Sending event at 0,1 state off
[00:12:59.781,219] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 1, position: 1, pressed: false
[00:12:59.781,311] <dbg> zmk: position_state_changed_listener: 14 capturing 1 up event
[00:12:59.781,372] <dbg> zmk: decide_hold_tap: 14 decided hold-interrupt (balanced decision moment other-key-up)
[00:12:59.781,402] <dbg> zmk: on_keymap_binding_pressed: position 14 keycode 0x700E0
[00:12:59.781,433] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0xE0 implicit_mods 0x00 explicit_mods 0x00
[00:12:59.781,433] <dbg> zmk: zmk_hid_register_mod: Modifier 0 count 1
[00:12:59.781,463] <dbg> zmk: zmk_hid_register_mod: Modifiers set to 0x01
[00:12:59.781,463] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x01
[00:12:59.781,494] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:12:59.781,585] <dbg> zmk: release_captured_events: Releasing key position event for position 1 pressed
[00:12:59.781,616] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
[00:12:59.781,646] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 1, binding name: KEY_PRESS
[00:12:59.781,707] <dbg> zmk: on_keymap_binding_pressed: position 1 keycode 0x70014
[00:12:59.781,738] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x14 implicit_mods 0x00 explicit_mods 0x00
[00:12:59.781,738] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x01
[00:12:59.781,799] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:12:59.781,951] <dbg> zmk: release_captured_events: Releasing key position event for position 1 released
[00:12:59.781,951] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
[00:12:59.782,043] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 1, binding name: KEY_PRESS
[00:12:59.782,073] <dbg> zmk: on_keymap_binding_released: position 1 keycode 0x70014
[00:12:59.782,104] <dbg> zmk: wpm_event_listener: key_pressed_count 1 keycode 20
[00:12:59.782,135] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x14 implicit_mods 0x00 explicit_mods 0x00
[00:12:59.782,135] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x01
[00:12:59.782,165] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:12:59.782,958] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
[00:12:59.783,325] <dbg> zmk: kscan_matrix_read: Sending event at 1,2 state off
[00:12:59.783,447] <dbg> zmk: zmk_kscan_process_msgq: Row: 1, col: 2, position: 14, pressed: false
[00:12:59.783,477] <dbg> zmk: position_state_changed_listener: 14 bubble (no undecided hold_tap active)
[00:12:59.783,508] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 14, binding name: HOMEROW_MODS
[00:12:59.783,599] <dbg> zmk: on_keymap_binding_released: position 14 keycode 0x700E0
[00:12:59.783,630] <dbg> zmk: wpm_event_listener: key_pressed_count 2 keycode 224
[00:12:59.783,630] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0xE0 implicit_mods 0x00 explicit_mods 0x00
[00:12:59.783,660] <dbg> zmk: zmk_hid_unregister_mod: Modifier 0 count: 0
[00:12:59.783,660] <dbg> zmk: zmk_hid_unregister_mod: Modifier 0 released
[00:12:59.783,660] <dbg> zmk: zmk_hid_unregister_mod: Modifiers set to 0x00
[00:12:59.783,691] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[... 28 lines cut ...]
[00:17:44.425,750] <dbg> zmk: on_keymap_binding_pressed: position 1 keycode 0x70014
[00:17:44.425,811] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x14 implicit_mods 0x00 explicit_mods 0x00
[00:17:44.425,811] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
[00:17:44.425,842] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:17:44.426,330] <dbg> zmk: on_keymap_binding_released: position 14 keycode 0x70016
[00:17:44.426,361] <dbg> zmk: wpm_event_listener: key_pressed_count 1 keycode 22
[00:17:44.426,361] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x16 implicit_mods 0x00 explicit_mods 0x00
[00:17:44.426,391] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:17:44.426,391] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
[00:17:44.427,246] <dbg> zmk: on_hold_tap_binding_released: 14 cleaning up hold-tap
[00:17:44.467,193] <dbg> zmk: kscan_matrix_read: Sending event at 0,1 state off
[00:17:44.467,376] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 1, position: 1, pressed: false
[00:17:44.467,407] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
[00:17:44.467,437] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 1, binding name: KEY_PRESS
[00:17:44.467,498] <dbg> zmk: on_keymap_binding_released: position 1 keycode 0x70014
[00:17:44.467,529] <dbg> zmk: wpm_event_listener: key_pressed_count 2 keycode 20
[00:17:44.467,529] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x14 implicit_mods 0x00 explicit_mods 0x00
[00:17:44.467,559] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
[00:17:44.467,559] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07

</details>

I'm not sure if this is considered a bug, but at least the behavior doesn't match the docs:

> If `global-quick-tap` is enabled, then `quick-tap-ms` will apply not only when the given hold-tap is tapped, but for any key tapped before it. This effectively disables the hold-tap when typing quickly, which can be quite useful for homerow mods.

urob, 2023-09-01:
This is to be expected. The global-quick-tap setting prevents hold-activations when the hold-tap is pressed within the timeout **after** another key. Your examples are about the hold-tap being pressed **before** another key.

The difference that you are observing is due to the `balanced` setting, which precisely triggers the hold when another key is both pressed and released while holding the hold-tap. If you don't want that, you could switch to the `tap-preferred` flavor instead.

danmou, 2023-09-01:
@urob ah thanks for clarifying. I would still like to find a way to get the behavior I was expecting though, since `global-quick-tap` won't have an effect If there's a slight pause before the problematic keystroke. Switching to tap-preferred won't work for me since I do want to retain the ability to trigger the modifier quickly. I guess what I actually want is either:
- A minimum time between pressing the mod-tap key and the next key, so if the next key is pressed too soon then a tap is triggered, or
- A minimum hold time, i.e. if the mod-tap key is only held very briefly it's always treated as a tap

In the example I pasted above the mod-tap key was held for less than 100 ms, which is much less than I would when intentionally using it as a modifier, so this shouldn't have any negative effects for me. Is there any way to achieve this?

urob, 2023-09-01:
I guess what I actually want is either:
> 
>     * A minimum time between pressing the mod-tap key and the next key, so if the next key is pressed too soon then a tap is triggered, or

There is currently no setting doing this.

> 
>     * A minimum hold time, i.e. if the mod-tap key is only held very briefly it's always treated as a tap
> 

This is what `tapping-term-ms` does *subject to exceptions defined by the flavor*. There is no way to define an exception-free tapping-term while simultaneously insisting on defining an exception such as `balanced`. (Doing so, would require two separate tapping-terms, one which can be over-ruled by flavor exceptions and one that can't).

Thinking a bit out of the box, I think your best bets are:

1. Lower your `tapping-term-ms` and go with the `tap-preferred` flavor: Setting the flavor to `tap-preferred` gets rid of the troubling exception, while lowering the  tapping-term addresses the concern that it prevents you from quickly firing mods.
2. Alternatively, increase your `quick-tap-ms` setting (150-200 are sensible settings) and keep setting the flavor to `balanced`. This way, you are less likely to end up in situations where the mod is fired off accidentally in the first place, allowing you to keep a more conservative tapping-term and the balanced flavor.

caksoylar, 2024-02-27:
Closing this one since the original issue seems to have been a misunderstanding, feel free to open a new issue or re-open and edit if you want to discuss this one specifically:
>     A minimum time between pressing the mod-tap key and the next key, so if the next key is pressed too soon then a tap is triggered, or

closed 2024-02-27 by caksoylar