zmkfirmware/zmk #1926: Update docs following directed advertising

JeppeKlitgaard, 2023-09-05:
Following the merge of #1913, the documentation should be updated.

Happy to do a PR for this soon.

Affected pages: https://zmk.dev/docs/behaviors/bluetooth#bluetooth-pairing-and-profiles

> If a profile has been paired but the peer is not connected yet, ZMK will also advertise itself as connectable. In the future, the behavior might change to direct advertising which only target the peer with the stored bluetooth MAC address. In this state, if the peer is powered on and moved within the distance of bluetooth signal coverage, it should automatically connect to the keyboard.

petejohanson, 2023-09-05:
> Following the merge of #1913, the documentation should be updated.
> 
> Affected pages: https://zmk.dev/docs/behaviors/bluetooth#bluetooth-pairing-and-profiles
> 
> > If a profile has been paired but the peer is not connected yet, ZMK will also advertise itself as connectable. In the future, the behavior might change to direct advertising which only target the peer with the stored bluetooth MAC address. In this state, if the peer is powered on and moved within the distance of bluetooth signal coverage, it should automatically connect to the keyboard.
> 
> I would be happy to do a PR for this, though I am not completely sure what the new behaviour is. How does one reset a 'pairing' with directed advertising?

We don't yet do directed advertising to hosts, only to centrals from a split peripheral. 

I have some work in progress for directed advertising to hosts, but it's still not even in a PR yet.

JeppeKlitgaard, 2023-09-05:
Ah, I see. Apologies for a bit of over-eagerness in that case! Feel free to close this in that case, unless documentation is needed to describe the central-peripheral connection.

caksoylar, 2023-09-06:
Closing this for now, since the central-peripheral connection shouldn't cause any user-facing changes.

closed 2023-09-06 by caksoylar