zmkfirmware/zmk #1929: Recommended way to build out-of-tree boards or shields?

jablko, 2023-09-07:
Is there documentation on how to build out-of-tree boards or shields?

I have three involved repositories, I think:
- https://github.com/zmkfirmware/zmk
- My [zmk-config](https://github.com/jablko/miryoku_zmk)
- The [out-of-tree shield](https://github.com/petejohanson/revxlp-config)

To build them, I added to my [config/west.yml](https://github.com/jablko/miryoku_zmk/blob/master/config/west.yml):
```yml
manifest:
  ...
  projects:
    - name: revxlp-config
      remote: petejohanson
      revision: HEAD
```

Then to my [build.yaml](https://github.com/jablko/miryoku_zmk/blob/master/build.yaml) I added:
```yml
include:
  - board: seeeduino_xiao_ble
    shield: revxlp
    cmake-args: -DBOARD_ROOT="${GITHUB_WORKSPACE}/revxlp-config/config"
```

It worked, but is this the recommended way/is there a better one? Is there documentation on how to build out-of-tree boards or shields, or on the `BOARD_ROOT` variable? Related: I found [this completed issue](https://github.com/zmkfirmware/zmk/issues/138) on documenting the `ZMK_CONFIG` variable.

Before I got it working I tried adding to my `config/west.yml`:
```yml
manifest:
  ...
  projects:
    - name: revxlp
      remote: petejohanson
      repo-path: revxlp-config/config/boards/shields/revxlp
      revision: HEAD
```

That failed: [fatal: repository 'https://github.com/petejohanson/revxlp-config/config/boards/shields/revxlp/' not found](https://github.com/jablko/miryoku_zmk/actions/runs/6099543069/job/16551958029#step:7:26). [West](https://docs.zephyrproject.org/latest/develop/west/manifest.html#projects) can't clone a repository's subdirectory?

In this case at least the out-of-tree repository is laid out just like the zmk repository (and my zmk-config repository, also), so the next thing I tried is cloning it over top of the zmk directory, by adding to my `config/west.yml`:
```yml
manifest:
  ...
  projects:
    - name: revxlp-config
      remote: petejohanson
      revision: HEAD
      path: zmk
```

That also failed: [Hint: project revxlp-config path "zmk" is taken by project zmk](https://github.com/jablko/miryoku_zmk/actions/runs/6099665759/job/16552352020#step:7:23). I didn't try cloning it over top of the root, zmk-config directory --- I doubt that's possible?

That's when I found the `BOARD_ROOT` and `cmake-args` variables --- is this the recommended way to build out-of-tree boards or shields?

caksoylar, 2023-12-11:
Building with boards/shields found in Zephyr modules is now officially supported via #1989. The setup script now creates a Zephyr module with a `boards` folder by default (see the [template repo](https://github.com/zmkfirmware/unified-zmk-config-template/)), but existing repos like https://github.com/petejohanson/revxlp-config will need to be updated to have `/zephyr/module.yml` and `/config/boards` folder moved to `/boards`.

Here is the updated documentation on [building with modules](https://zmk.dev/docs/development/build-flash#building-with-external-modules) and the [new shield guide](https://zmk.dev/docs/development/new-shield#new-zephyr-module-repository) that encourages defining new shields as modules.

nmunnich, 2025-08-14:
I believe https://zmk.dev/docs/features/modules suffices for documentation here now.

closed 2025-08-14 by nmunnich