zmkfirmware/zmk #1931: [Feature Request] Support Multiple Keymaps For Same Board & Shield Combination

feraleyebrows, 2023-09-08:
It would be nice to be able to be able to define multiple keymaps for the same keyboard, especially when you're trying to dial in a new keymap and want to try a few different layouts. I know I can define one, push it, download the artifact and save it somewhere then repeat the process or build locally using the -DZMK_CONFIG setting but it would be nicer if this could be exposed in the build.yaml file. E.g

```
include:
  - board: nice_nano_v2
    shield: cradio_left
    keymap: cradio-01
  - board: nice_nano_v2
    shield: cradio_right
    keymap: cradio-01
  - board: nice_nano_v2
    shield: cradio_left
    keymap: cradio-02
  - board: nice_nano_v2
    shield: cradio_right
    keymap: cradio-02
```
or preferably:
```
include:
  - board: nice_nano_v2
    shield: cradio_left
    - keymaps: 
      - cradio-01
      - cradio-02
  - board: nice_nano_v2
    shield: cradio_right
    keymaps:
      - cradio-01
      - cradio-02
```

Then inside zmk-config/config you would have the following files:
cradio.config
cradio-01.keymap
cradio-02.keymap

caksoylar, 2023-12-11:
This is now possible with `artifact-name` property in combination with `cmake-args` to specify an alternate keymap, since #2015. Here is an [example](https://github.com/caksoylar/zmk-config/blob/test-keymap/build.yaml): 
```yaml
include:
  - board: nice_nano_v2
    shield: corne_left
  - board: nice_nano_v2
    shield: corne_left
    cmake-args: -DKEYMAP_FILE=../../config/corneish_zen.keymap
    artifact-name: corne_left_with_zen_keymap
```

This isn't the most intuitive since `KEYMAP_FILE` isn't documented for end users, and the path is relative to `zmk/app` so it requires the additional `../../`. It might be good to support a `keymaps` property officially.

jonathan-elize, 2024-02-11:
Hi @caksoylar is there a way to get this to work with different .conf files? I have two corne keyboards and I would like them each to have their own separate names. Trying to do the following is leading to build errors when I try and compile:

```yaml
include:
  - board: nice_nano_v2
    shield: corne_left
    cmake-args: -DCONFIG_ZMK_KEYBOARD_NAME=mawon
    artifact-name: mawon_corne_left
  - board: nice_nano_v2
    shield: corne_right
    artifact-name: mawon_corne_right
    cmake-args: -DCONFIG_ZMK_KEYBOARD_NAME=mawon
  - board: nice_nano_v2
    shield: settings_reset
  - board: nice_nano_v2
    shield: corne_left
    cmake-args: -DCONFIG_ZMK_KEYBOARD_NAME=noir -DCONFIG_ZMK_RGB_UNDERGLOW=y
    artifact-name: noir_corne_left
  - board: nice_nano_v2
    shield: corne_right
    cmake-args: -DCONFIG_ZMK_KEYBOARD_NAME=noir -DCONFIG_ZMK_RGB_UNDERGLOW=y
    artifact-name: noir_corne_right
```

caksoylar, 2024-02-11:
@jonathan-elize I think the keyboard name config, being a string type, needs quotation marks around the value. But if you just surround them with double quotes I don't know if they would survive yaml+shell script. Maybe try `cmake-args: '-DCONFIG_ZMK_KEYBOARD_NAME="noir"' -DCONFIG_ZMK_RGB_UNDERGLOW=y`, see if that works.

For .conf, `-DEXTRA_CONF_FILE=` might work (to append a new conf file), but I am not sure if it breaks anything or would override settings like you'd expect it to.

jonathan-elize, 2024-02-11:
@caksoylar  Thanks for the quick reply! I tried using the the name with quotes around it but I still received the following error: `/tmp/tmp.pwHopIq3Sk/zephyr/misc/generated/extra_kconfig_options.conf:2: warning: malformed string literal in assignment to ZMK_KEYBOARD_NAME (defined at /tmp/tmp.YtWFEy9DG5/zmk/app/Kconfig:10, `

I also tried using the following config but it seemed to have no affect:
```yaml
include:
  - board: nice_nano_v2
    shield: corne_left
    artifact-name: mawon_corne_left
  - board: nice_nano_v2
    shield: corne_right
    artifact-name: mawon_corne_right
  - board: nice_nano_v2
    shield: settings_reset
  - board: nice_nano_v2
    shield: corne_left
    cmake-args: -DEXTRA_CONF_FILE=../../config/corne-noir.conf
    artifact-name: noir_corne_left
  - board: nice_nano_v2
    shield: corne_right
    cmake-args: -DEXTRA_CONF_FILE=../../config/corne-noir.conf
    artifact-name: noir_corne_right
```

The structure of my directory is:

```
 zmk-corne/
   - boards/
   - config/
     - corne-noir.conf
     - corne.conf
     - corne.keymap
     - west.yml
   - zephyr/
   - build.yaml
```

caksoylar, 2024-02-12:
Looks like the quoting didn't survive, like I suspected: https://github.com/jonathan-elize/zmk-corne/actions/runs/7864892654/job/21457196159#step:11:1
It might be possible that some combination of backslashes and quoting would get it there but it is likely to be too much trouble to be intuitive.

The extra conf flag was also not respected for some reason https://github.com/jonathan-elize/zmk-corne/actions/runs/7864924611/job/21457274795#step:11:78, I didn't have that warning when testing locally. But I didn't have a `config/corne.conf` file, maybe that has an impact.

Multiple branches seem like the easiest way to do it I suppose, unless you figure out the quoting/escaping mess for `cmake-args`.

thebino, 2024-03-19:
@mardybardy I have had the same and solved it with a different path in the GitHub workflow: https://github.com/thebino/zmk-config

chelming, 2024-06-17:
It _looks_ like escaped double quotes should do it based on https://github.com/zephyrproject-rtos/zephyr/pull/49267 but... it's not: 
https://github.com/chelming/zmk-config/actions/runs/9552363370/job/26328777346#step:11:13

It builds without errors, it just uses the "stock" shield.

caksoylar, 2024-06-17:
The build you linked manages to change the keyboard name properly: https://github.com/chelming/zmk-config/actions/runs/9552363370/job/26328777346#step:12:747

Note that setting the keyboard name will not make it pick up a different shield/keymap. You need to use the `KEYMAP_FILE` arg that I noted above.

chelming, 2024-06-17:
@caksoylar thanks, you're right! :heart: 

so, it looks like this works as expected:
```yaml
include:
  - board: nice_nano_v2
    shield: corne_left nice_view_adapter nice_view
    cmake-args: -DKEYMAP_FILE=../../config/swept-corne.keymap -DCONFIG_ZMK_KEYBOARD_NAME=\"swept-corne\"
    artifact-name: swept_corne_left
 ```

chelming, 2024-06-17:
just tested a separate build conf and that works too. I think everything is working now.

build.yaml:
```yaml
include:
  - board: nice_nano_v2
    shield: corne_left nice_view_adapter nice_view
    cmake-args: -DKEYMAP_FILE=../../config/swept-corne.keymap -DCONFIG_ZMK_KEYBOARD_NAME=\"swept-corne\" -DEXTRA_CONF_FILE=../../config/swept-corne.conf
    artifact-name: swept_corne_left
  - board: nice_nano_v2
    shield: corne_right nice_view_adapter nice_view
    cmake-args: -DKEYMAP_FILE=../../config/swept-corne.keymap -DCONFIG_ZMK_KEYBOARD_NAME=\"swept-corne\" -DEXTRA_CONF_FILE=../../config/swept-corne.conf
    artifact-name: swept_corne_right
  - board: nice_nano_v2
    shield: settings_reset
```

MatchstickWorks, 2024-07-20:
Wanted to drop a thank you for this. My file structure is a little wonky, but I got this working by adding a keymap to `config` and leaving my default keymap under `boards/shields/...`

I don't exactly know why it doesn't work if I point it to the location of my default keymap, but this works.

[Here's my config](https://github.com/MatchstickWorks/zmk-config/tree/master) if anyone would like another example.