zmkfirmware/zmk #1950: [Feature Request] Don't clear current profile unless BT_CLR is held for X seconds

muskwasis, 2023-10-04:
Too many times when flailing do I accidentally clear my profile and the keyboard disconnects. Could we add a delay or another command BT_CLR_XS? Where X is a delay in seconds.

caksoylar, 2023-10-04:
Tap dance is a good idea, another option would be to put it in a hold-tap, see documentation (along with the notes in the info box) at https://zmk.dev/docs/behaviors/hold-tap#using-different-behavior-types-with-hold-taps.

Something like this (not tested):
```dts
// wrapper macro needed for placing it in a hold-tap
ZMK_MACRO(bt_clr_macro, bindings = <&bt BT_CLR>;)

// hold-tap that doesn't do anything on tap and BT_CLR on hold
bt_clr_hold: bt_clr_hold {
    compatible = "zmk,behavior-hold-tap";
    label = "bt_clr_hold";
    #binding-cells = <2>;
    flavor = "tap-preferred";
    tapping-term-ms = <500>;  // delay amount in ms
    bindings = <&bt_clr_macro>, <&none>;
};

// use it in your keymap like `&bt_clr_hold 0 0`
```

idesignstuff, 2023-10-05:
I'd also appreciate a &settings_reset command that would allow us to more quickly troubleshoot pairing halves, clearing bt settings etc. Imagine a world where we aren't compiling the same code to be flashed/reset/flashed/reset with every try?

joelspadin, 2023-10-05:
> I'd also appreciate a &settings_reset command that would allow us to more quickly troubleshoot pairing halves, clearing bt settings etc. Imagine a world where we aren't compiling the same code to be flashed/reset/flashed/reset with every try?

#1757 adds a way to do this.