zmkfirmware/zmk #1977: Weak signal between split keyboard halves

szczys, 2023-10-25:
I have a corne keyboard using nice!nano v2 boards (nrf52840). They work great on the desk, but if I put the halves on either side of my legs when sitting on the sofa, the connection is spotty and often drops.

I have used the `CONFIG_BT_CTLR_TX_PWR_PLUS_8=y` symbol to increase the power. This helps stabilize the connection with the computer but I don't think it has improved the connection between the halves.

Is there anything I can do to stabilize the signal?

caksoylar, 2023-10-25:
`CONFIG_BT_CTLR_TX_PWR_PLUS_8=y` will also affect the signal between halves if it affects the right half, e.g. if it is in `config/corne.conf` or `config/corne_right.conf` (if you don't have a `config/corne.conf` file). That's the only solution in firmware side. Unrfortunately human body is a good blocker for wireless signals.

szczys, 2023-10-25:
Thank you, I'll try adding that to the right half conf and see if it solves my issue.

closed 2023-10-25 by szczys

tomgroenwoldt, 2024-02-14:
See my comment [here](https://github.com/zmkfirmware/zmk/issues/916#issuecomment-1943816527). This solved my connection issues when having both halves far apart from each other.