zmkfirmware/zmk #1999: C_VOL_UP and C_VOL_DOWN not working on Android

hvuitsme, 2023-11-11:
default_layer {
            label = "DF";

            sensor-bindings = <&inc_dec_kp C_VOL_UP C_VOL_DN>;

            bindings = <
                &mo 1           &kp LEFT        &kp UP            &kp DOWN        &kp RIGHT       &kp F5
                &trans          &trans          &trans            &trans          &trans          &trans
                &trans          &trans          &trans            &trans          &trans          &trans
                &kp C_MUTE
            >;
        };

Currently when I use rotary encoder it doesn't work but when I change it from C_ to K_ it works.
As for keycodes related to C_, they do not work with most Android devices

caksoylar, 2023-11-11:
Can you confirm whether the C_ keycodes work on a layer, rather than on the sensor bindings?

hvuitsme, 2023-11-12:
> Can you confirm whether the C_ keycodes work on a layer, rather than on the sensor bindings?

sensor-bindings = <&inc_dec_kp C_VOL_UP C_VOL_DN>;

            bindings = <
                &mo 1           &kp LEFT        &kp UP            &kp C_VOL_UP    &kp C_VOL_UP    &kp F5
                &kp C_AC_BACK   &kp SPACE       &kp LS(N)         &trans          &trans          &trans
                &trans          &trans          &trans            &trans          &trans          &trans
                &kp C_MUTE
            >; 

-1 : as above, I tried with 2 C_VOL_UP on columns 4 and 5 and the result was that the two buttons worked normally and the encoder worked normally.
-2 : I tried deleting 1 C_VOL_UP and replacing it with C_VOL_DN, it did not work on both buttons of columns 4 and 5 and the encoder also did not work (except for android, the rest worked normally)

hvuitsme, 2023-11-12:
My entire zmkconfig is here: https://github.com/hvuitsme/test_zmk

caksoylar, 2023-11-12:
> -1 : as above, I tried with 2 C_VOL_UP on columns 4 and 5 and the result was that the two buttons worked normally and the encoder worked normally.
> -2 : I tried deleting 1 C_VOL_UP and replacing it [...]

This sounds like a random effect to me, whether you have `C_VOL_DN` in some other location shouldn't affect how `C_VOL_UP` works.

Overall I am not sure why, but it sounds pretty random and maybe not specific to encoder behavior given 2. Maybe Android is more strict with key timings for that keycode and connection quality affects whether it works or not.

hvuitsme, 2023-11-12:
> > -1 : như trên mình thử với 2 C_VOL_UP ở cột 4 và 5 thì kết quả là 2 nút đó hoạt động bình thường và bộ mã hóa hoạt động bình thường. 
> > -2 : Mình thử xóa 1 C_VOL_UP và thay bằng C_VOL_DN thì nó không hoạt động ở cả 2 nút cột 4 và 5 và bộ mã hóa cũng không hoạt động (ngoại trừ android, còn lại hoạt động bình thường)
> 
> Đối với tôi, điều này nghe có vẻ như là một hiệu ứng ngẫu nhiên, cho dù bạn có `C_VOL_DN`ở một địa điểm nào khác cũng không ảnh hưởng đến cách thức `C_VOL_UP`hoạt động.
> 
> Nhìn chung, tôi không chắc tại sao, nhưng nó nghe có vẻ khá ngẫu nhiên và có thể không cụ thể đối với hành vi của bộ mã hóa được đưa ra 2. Có thể Android nghiêm ngặt hơn với thời gian chính cho mã khóa đó và chất lượng kết nối ảnh hưởng đến việc nó có hoạt động hay không.

I also don't know much about this situation. I have tried many people now and their situation is similar, sometimes it works, sometimes it doesn't.

hvuitsme, 2023-11-12:
> This sounds like a random effect to me, whether you have `C_VOL_DN` in some other location shouldn't affect how `C_VOL_UP` works.
>
> Overall I am not sure why, but it sounds pretty random and maybe [...]

I also don't know much about this situation. I have tried many people now and their situation is similar, sometimes it works, sometimes it doesn't.