zmkfirmware/zmk #2001: [BUG] Mod-taps getting stuck

w0uld, 2023-11-11:
Reviving issue #986 | I've encountered a very similar issue, but instead of combos, it's paired with layer switching. 

Long story short: 
- Quick layer switches and mod-taps don't always work well together. I had accented keybinds on a secondary layer that I used fairly frequently, and when I used that layer with a shift/equal mod-tap, on release I sometimes had shift stuck until I pressed it again. It occurs when I trigger & release both the layer switch and the mod-tap's hold behavior in a short timespan. 
- I decided to switch to mod-morphs to see if the issue persisted. With mod-morphs instead of layer switches, the issue is non-existent. This is my workaround for the time being. 
  
Note that this issue persists across multiple keyboards with the same config: Corne-ish Zen & Choctopus44. I originally thought that this was a sync issue between the 2 halves of the Zen, but the Choctopus being a unibody board indicates an issue potentially on the firmware side.

Log after pressing & releasing layer switch, mod-tap, and a third key (to trigger the hold behavior) in a very short timeframe, causing the issue to occur: 

> [00:07:19.739,654] <dbg> zmk: peripheral_event_work_callback: Trigger key position state change for 40
> [00:07:19.739,685] <dbg> zmk: position_state_changed_listener: 40 bubble (no undecided hold_tap active)
> [00:07:19.739,776] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 40, binding name: STICKY_LAYER
> [00:07:19.739,807] <dbg> zmk: on_sticky_key_binding_released: Another key was pressed while the sticky key was pressed. Act like a normal key.
> [00:07:19.739,837] <dbg> zmk: mo_keymap_binding_released: position 40 layer 3
> [00:07:19.739,868] <dbg> zmk: set_layer_state: layer_changed: layer 3 state 0
> [00:07:19.773,162] <dbg> zmk: kscan_matrix_read: Sending event at 0,1 state off
> [00:07:19.773,345] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 1, position: 1, pressed: false
> [00:07:19.773,376] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
> [00:07:19.773,437] <dbg> zmk: zmk_keymap_apply_position_state: layer: 3 position: 1, binding name: KEY_PRESS
> [00:07:19.773,468] <dbg> zmk: on_keymap_binding_released: position 1 keycode 0x7006F
> [00:07:19.773,529] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x6F implicit_mods 0x00 explicit_mods 0x00
> [00:07:19.773,529] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
> [00:07:19.773,559] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
> [00:07:20.424,163] <dbg> zmk: kscan_matrix_read: Sending event at 0,1 state on
> [00:07:20.424,316] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 1, position: 1, pressed: true
> [00:07:20.424,377] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
> [00:07:20.424,407] <dbg> zmk: position_state_down: combo: capturing position event 1
> [00:07:20.424,438] <dbg> zmk: zmk_event_manager_handle_from: Listener captured the event
> [00:07:20.444,519] <dbg> zmk: filter_timed_out_candidates: after filtering out timed out combo candidates: remaining_candidates=1 timestamp=440444
> [00:07:20.474,639] <dbg> zmk: filter_timed_out_candidates: after filtering out timed out combo candidates: remaining_candidates=0 timestamp=440474
> [00:07:20.474,670] <dbg> zmk: release_pressed_keys: combo: releasing position event 1
> [00:07:20.474,731] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 1, binding name: KEY_PRESS
> [00:07:20.474,761] <dbg> zmk: on_keymap_binding_pressed: position 1 keycode 0x70034
> [00:07:20.474,822] <dbg> zmk: hid_listener_keycode_pressed: usage_page 0x07 keycode 0x34 implicit_mods 0x00 explicit_mods 0x00
> [00:07:20.474,822] <dbg> zmk: zmk_hid_implicit_modifiers_press: Modifiers set to 0x00
> [00:07:20.474,853] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07
> [00:07:20.502,166] <dbg> zmk: kscan_matrix_read: Sending event at 0,1 state off
> [00:07:20.502,288] <dbg> zmk: zmk_kscan_process_msgq: Row: 0, col: 1, position: 1, pressed: false
> [00:07:20.502,349] <dbg> zmk: position_state_changed_listener: 1 bubble (no undecided hold_tap active)
> [00:07:20.502,410] <dbg> zmk: zmk_keymap_apply_position_state: layer: 0 position: 1, binding name: KEY_PRESS
> [00:07:20.502,441] <dbg> zmk: on_keymap_binding_released: position 1 keycode 0x70034
> [00:07:20.502,502] <dbg> zmk: hid_listener_keycode_released: usage_page 0x07 keycode 0x34 implicit_mods 0x00 explicit_mods 0x00
> [00:07:20.502,502] <dbg> zmk: zmk_hid_implicit_modifiers_release: Modifiers set to 0x00
> [00:07:20.502,532] <dbg> zmk: zmk_endpoints_send_report: usage page 0x07

caksoylar, 2023-11-11:
Can you note the exact order of key presses and releases and position of the key for the [keymap that you linked](https://github.com/w0uld/zmk-config-zen-2/blob/f9cf21d3ead678465b7e87c8bd44209ac61950bb/config/corneish_zen.keymap)? I had trouble understanding how your accent layer involves mod-taps. As a bonus, that would make it much easier to create a [test](https://zmk.dev/docs/development/tests) that would reproduce the issue, which in turn makes it easier to debug and fix.

w0uld, 2023-11-11:
The keys involved are: 
1. &sl ACNT (right thumb cluster middle key)
2. &mt LSHFT EQUAL (bottom row rightmost key)
3. any of the ACNT layer's F15-F23 keys on the left hand side

Order of pressing: 1, 2, 3
Order of release: can't tell exactly, the issue occurs only if I release them almost immediately.

caksoylar, 2023-11-11:
That is helpful, thanks.

If you wanted to go the extra mile, you could try writing a test like I mentioned. What I usually try to do is reproduce the issue, then start simplifying the setup until it no longer reproduces. Simplifying involves e.g. removing combos, reducing the layer size to only contain the necessary keys, remove extra properties from behaviors, etc. This is much easier to do with a test since you don't have to flash your keyboard everytime and try to reproduce press/release timings manually.

w0uld, 2023-11-11:
Looking at the description it doesn't look like I'd be able to actually run the test. I'm on Windows, and reading the guide I'd need to set up a Linux environment to do this.

caksoylar, 2023-11-11:
I see, I use WSL2 on Windows with Ubuntu/Debian for ZMK work, but it is up to you whether you want to invest in it or not.

w0uld, 2023-11-11:
That part is somewhat over my level of expertise. I think I'll pass on that. :/