zmkfirmware/zmk #2009: Last commit is breaking github workflow

dsych, 2023-11-13:
Hey,
I was in a process of updating a keymap for my custom split board though github actions and started seeing compilation errors all of the a sudden related to usb_hid. See below:

```
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /__w/zmk-dev/zmk-dev/build/zephyr/zephyr_pre0.map 
: && ccache /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -Wl,-T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/__w/zmk-dev/zmk-dev/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a  zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a  zephyr/arch/arch/arm/core/aarch32/mpu/libarch__arm__core__aarch32__mpu.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/soc/arm/common/cortex_m/libsoc__arm__common__cortex_m.a  zephyr/soc/arm/nordic_nrf/nrf52/libsoc__arm__nordic_nrf__nrf52.a  zephyr/subsys/bluetooth/common/libsubsys__bluetooth__common.a  zephyr/subsys/bluetooth/host/libsubsys__bluetooth__host.a  zephyr/subsys/bluetooth/controller/libsubsys__bluetooth__controller.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/usb/device/class/hid/libsubsys__usb__device__class__hid.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/usb/device/libdrivers__usb__device.a  zephyr/drivers/adc/libdrivers__adc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/sensor/nrf5/libdrivers__sensor__nrf5.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/kscan/libdrivers__kscan.a  zephyr/drivers/flash/libdrivers__flash.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/timer/libdrivers__timer.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/module/drivers/sensor/battery/lib..__zmk__app__module__drivers__sensor__battery.a  modules/module/lib/zmk_debounce/lib..__zmk__app__module__lib__zmk_debounce.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/thumb/v7e-m/nofp"  -L/__w/zmk-dev/zmk-dev/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -no-pie  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn && cd /__w/zmk-dev/zmk-dev/build/zephyr && /usr/local/lib/python3.8/dist-packages/cmake/data/bin/cmake -E echo
/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(usb_hid.c.obj): in function `get_keyboard_report':
/__w/zmk-dev/zmk-dev/zmk/app/src/usb_hid.c:43: undefined reference to `zmk_hid_get_keyboard_report'
/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /__w/zmk-dev/zmk-dev/zmk/app/src/usb_hid.c:48: undefined reference to `zmk_hid_get_boot_report'
/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(usb_hid.c.obj): in function `get_report_cb':
/__w/zmk-dev/zmk-dev/zmk/app/src/usb_hid.c:73: undefined reference to `zmk_hid_get_consumer_report'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /__w/zmk-dev/zmk-dev/build
Error: Process completed with exit code 1.
```

I dug a bit and it turns out [the latest commit](https://github.com/zmkfirmware/zmk/commit/91aa3378f32f5688329a663fdaba4772e069d548) added the exact line mentioned in the error message. 

I am not an expert on GH Actions, but it appears to me that every GH workflow relying on the tip on `main` is going to fail until this is addressed. 

In the meantime, do you know how I can pin the exact commit id which zmk's GH build file is using? I skimmed [build-user-config.yml](https://github.com/zmkfirmware/zmk/blob/main/.github/workflows/build-user-config.yml), but there appear to be no way for me to pass the commit id for checkout...

What's weird is this error is coming up only for the `right` half and the left is building just fine. I tried disabling the `usb_hid` via `ZMK_USB_BOOT=n` to no avail. 

Cheers!

petejohanson, 2023-11-13:
Please try with #2006 and report back. I opened that today to specifically fix this issue that was revealed by the USB boot merge.

petejohanson, 2023-11-13:
In the meantime, you can update `west.yml` file in your config repo to pin if you need.

J4ckTh3R1qp3r, 2023-11-13:
also have problems with github actions
```
Run mkdir build/artifacts
+ mkdir build/artifacts
+ [ -f build/zephyr/zmk.uf2 ]
+ cp build/zephyr/zmk.uf2 build/artifacts/splitkb_aurora_corne_right nice_view_adapter nice_view-nice_nano_v2-zmk.uf2
cp: target 'nice_view-nice_nano_v2-zmk.uf2' is not a directory
Error: Process completed with exit code 1.
```

caksoylar, 2023-11-13:
> also have problems with github actions
>
> ```
> Run mkdir build/artifacts
> + mkdir build/artifacts
> + [ -f build/zephyr/zmk.uf2 ]
> + cp build/zephyr/zmk.uf2 build/artifacts/splitkb_aurora_corne_right [...]

This is on me through #2004, apparently I missed a test case. I'll try to open a PR to fix, or we can revert #2004 until we can.

petejohanson, 2023-11-13:
Reverted for now, until @caksoylar can get a chance to address in a follow up PR. Thanks!

dsych, 2023-11-14:
tried the head of `main`. works like a charm! thanks!

closed 2023-11-14 by dsych