zmkfirmware/zmk #2012: Copying uf2 for firmware using encoders on peripheral side fails

bzwartsenberg, 2023-11-14:
Let me start by saying what an amazing project this is, I have just started to see some bits of it and  it is really cool what this firmware is able to do!

I am trying to set up a custom keyboard, and I am using the sofle setup for testing purposes. I am trying to get an encoder working on the peripheral side. I set up a user configuration repo using the bash script, and compile it using the CLI. The only modification I made to the auto generated user repo is uncommenting the lines that add encoder support. I.e. my `sofle.conf` file looks like this:
```
...
# Uncomment these two lines to add support for encoders
CONFIG_EC11=y
CONFIG_EC11_TRIGGER_GLOBAL_THREAD=y
...
```
The code builds just fine, but when I go to flash the peripheral side, when copying the uf2 I get 

```
The operation can’t be completed because one or more required items can’t be found.
(Error code -43)
```
(This is a message from  Finder, on Mac OS, but something similar comes up when I use rsync or cp)

 - I tried running without the encoders and both halves flash just fine, and the setup works as expected.
 - I tried compiling firmware with only an encoder on the left, using the `sofle_left.conf`, which works as expected. Both halves flash fine, the encoder on the left works, but the one on the right it does not.

I am probably doing something silly, but it seems very strange to me that the code compiles just fine but flashing the firmware does not work.

caksoylar, 2023-11-14:
Usually a firmware configuration change like that one will not cause flashing behavior to change. You can expect the keyboard to not work properly, but not for flashing to fail. I guess you never know what is happening with these clones but I would maybe try flashing from a different OS.

bzwartsenberg, 2023-11-14:
OK that's what I figured, thanks for confirming that in theory different configurations should not lead flashing to fail. I'll try a different OS and see if I have more luck, thanks!

bzwartsenberg, 2023-11-15:
I tried again using `cp -X`, and now it flashes OK and the encoder works! This must be something weird going on between Mac OS and the uf2 magic. Definitely not an issue with zmk.

closed 2023-11-15 by bzwartsenberg