zmkfirmware/zmk #2019: Question: Caret key on German keyboard (QUERTZ)

davidmondok, 2023-11-14:
I'm only asking because after spending a whole day without a solution I'm getting desperate.

I can't find a way to configure an Advantage 360 on my MacOS with the locale set to German to output the top left key (caret and degree sign). What would be the correct configuration to achieve that? 

[image]

caksoylar, 2023-11-14:
You can run [zmk-locale-generator](https://github.com/joelspadin/zmk-locale-generator) with the [OS X DE keyboard](https://github.com/unicode-org/cldr/blob/a88558301989b98d5bedb13e9123ec6d12f82416/keyboards/osx/de-t-k0-osx.xml):
```sh
python -m zmk_locale_generator -o keys_de_osx.h -v DE cldr/keyboards/osx/de-t-k0-osx.xml
```
The resulting header contains these:
```c
/* ^ */
#define DE_CARET (ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_NON_US_BACKSLASH_AND_PIPE))
...
/* ° */
#define DE_DEGREE (RC(ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_NON_US_BACKSLASH_AND_PIPE)))
#define DE_DEG (DE_DEGREE)
```
So it sounds like it should be `NUBS` for caret and `RC(NUBS)` for degree.

davidmondok, 2023-11-14:
Thank you very much @caksoylar ! Would you be so kind and tell me how the header information you generated with zmk-locale-generator lead you to `NUBS`? I'd love to understand that part so I can use this tool myself effectively. Right now I'm absolutely clueless how you made that conclusion :)

caksoylar, 2023-11-14:
You can map those strings back to their user-friendly variants by checking the keys header here: https://github.com/zmkfirmware/zmk/blob/main/app/include/dt-bindings/zmk/keys.h

For that one specifically, `HID_USAGE_KEY_KEYBOARD_NON_US_BACKSLASH_AND_PIPE` sounded like the "non-US backslash" key from here https://zmk.dev/docs/codes#symbols--punctuation to me so I just checked there.

davidmondok, 2023-11-15:
Thank you so much @caksoylar

I'll look into this more deeply, but just wanted to let you know, that `NUBS` was exactly what I was looking for :)

closed 2023-11-15 by davidmondok