zmkfirmware/zmk #2026: Can't pair to both Ubuntu and Windows on the same laptop.

BenBlumer, 2023-11-17:
So -- I dual boot Ubuntu and Windows 11. I had my keyboard paired with W11 (as well as a phone and a tablet). I then booted in Ubuntu and tried to pair. I was getting authorization failed messages from bluetoothctl. I cleared all the bluetooth profiles. It paired and connected. I then booted W11 and tried to re-pair. It wouldn't do it. This time, I only cleared the profile for the Ubuntu connection. All of a sudden, it would pair.

It seems like ZMK doesn't want to pair to two OSes on the same device (or with the same bluetooth MAC address?).

Is this a fundamental bluetooth issue or a ZMK issue?

_Originally posted by @BenBlumer in https://github.com/zmkfirmware/zmk/issues/1189#issuecomment-1815744862_

ReFil, 2023-11-17:
Zmk enforces strict security and won't store multiple keys for the same device, there is a way to do it though
https://hoelter.prose.sh/kinesis-advantage-360-bluetooth-dualboot

caksoylar, 2023-11-18:
I think it would be good to put a pointer to the wiki (with a short summary) in the docs, under the Bluetooth feature page troubleshooting header.

BenBlumer, 2023-11-19:
> Zmk enforces strict security and won't store multiple keys for the same device, there is a way to do it though https://hoelter.prose.sh/kinesis-advantage-360-bluetooth-dualboot

What is the security risk?  Genuinely asking. 

I can understand preventing a device from pairing itself by saying "Trust me -- we've already paired!". Is that risk not solved by insisting each OS must pair to a different BT profile on the ZMK? That way a user has to explicitly pair the device, as they always do, and we should have the same amount of security we always have.

piit79, 2025-03-06:
> What is the security risk? Genuinely asking.

Does anyone know? I'd love to understand what the supposed security risk is. Having to copy over the pairing keys feels like a significant low-level hassle that a regular user should not have to go through...

Would it be able to patch this behaviour out, even unofficially?j I have no idea at what part of the ZMK source to look at...

VegaDeftwing, 2026-08-16:
This definitely needs the Arch Wiki link replaced by the page @ReFil linked. That guide worked, trying to do it from the disaster that is the info on that Arch Wiki page made we want to commit crimes.