zmkfirmware/zmk #2040: `hold-trigger-on-release` cancels effect of `hold-trigger-key-positions` with `flavor = "hold-preferred";`

whxvd, 2023-11-27:
A week ago I got a Glove80. I am new to ZMK. I am currently trying to get a behavior like the following to work:

```
rmod: rmod {
  label           = "/behaviors/rmod";
  compatible      = "zmk,behavior-hold-tap";
  #binding-cells  = <2>;
  flavor          = "hold-preferred";
  tapping-term-ms = <1000>;
  bindings        = <&kp>, <&kp>;
  hold-trigger-key-positions = <LEFT_HAND_KEYS THUMB_KEYS>;
  hold-trigger-on-release;
};
```

But as soon as I add the `hold-trigger-on-release`, the effect of `hold-trigger-key-positions` seems to be nullified. It seems to be as if `hold-trigger-key-positions` were not specified.

Everything works as expected when I choose `flavor = tap-preferred`.

Until now, I only tested with the MoErgo ZMK repository. I also learned that there is a native posix target, with which I may be able to produce a minimal test against this repository. However, before I go down that route, I would like to have the following question answered:

Is `hold-trigger-on-release` supposed to work with `flavor = "hold-preferred"`? The documentation does not state any limitation.

caksoylar, 2023-11-27:
The `hold-trigger-on-release` property was added to be used with the `balanced` flavor, and then augmented to work well with `tap-preferred` as well. I don't know if any special considerations are needed for it to work well with `hold-preferred` -- @urob do you have an idea?

Conceptually, if you add the `on-release` check then does `hold-preferred` flavor become equivalent to `balanced`? The difference seems to me thate the release of the interrupting key doesn't need to happen within the tapping term.

whxvd, 2023-12-27:
> What would be the desired effect of setting `hold-trigger-on-release` with `hold-preferred`?

I intended to try `hold-preferred` for home-row mod shift keys. At least for me, that can only reasonably work with `hold-trigger-key-positions` (left/right) and `hold-trigger-on-release`. The latter is necessary, because otherwise I cannot produce, e.g., Shift+Control together: Without `hold-trigger-on-release`, pressing the home-row keys for Shift and Control together immediately triggers the corresponding taps.

When I got the new keyboard and started to experiment with new layouts, I was awfully slow at typing. After having "recovered" to a large extend, it has become clear that `hold-preferred` almost certainly never will work for me on the home-row. So this issue is not an issue for me personally, anymore.

However, the bug that `hold-trigger-on-release` seems to cancel a `hold-trigger-key-positions` setting remains.

I am not 100% sure whether the originally desired effect of a `hold-preferred` home-row mod shift that can be combined with other home-row modifiers (of different flavor) is in principle possible / well-defined at all. In the case of `hold-preferred`, it may be necessary or desirable to have `hold-trigger-on-release` to also mean the release of the modifier, and not only the release of the modified key. But I don't know.