zmkfirmware/zmk #2059: build-user-config action no longer copy parent directories

TristanCacqueray, 2023-12-06:
It seems like since https://github.com/zmkfirmware/zmk/pull/1989 , config using https://github.com/urob/zmk-nodefree-config no longer build because of such errors:

```
 /tmp/tmp.vVFsbR0CcL/config/corne.keymap:17:10: fatal error: ../zmk-nodefree-config/keypos_def/keypos_42keys.h: No such file or directory
   17 | #include "../zmk-nodefree-config/keypos_def/keypos_42keys.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

vvision, 2023-12-07:
Reproduced on my side today too.
```
/tmp/tmp.UUyIFrP4xQ/config/splitkb_aurora_corne.keymap:13:10: fatal error: ../src/corne.dtsi: No such file or directory
   13 | #include "../src/corne.dtsi"
      |          ^~~~~~~~~~~~~~~~~~~
```
I'm also gonna pin the version of `zmkfirmware/zmk/.github/workflows/build-user-config.yml`.

petejohanson, 2023-12-07:
Can you give this a try?

https://github.com/petejohanson/revxlp-config/actions/runs/7132623640/workflow

Doing so will *only* copy if we detect the new Zephyr module bits, so we shouldn't break existing configs with that change. If it works for you I'll get this into a PR.

ObiWanKeoni, 2023-12-07:
@petejohanson that seems to [work](https://github.com/ObiWanKeoni/zmk-config/actions/runs/7135073855) for my config!

lou-bi, 2023-12-08:
Hi, got this exact issue while building my firmware with Urob (not as a submodule).
I've follow your link @petejohanson , but Im not sure I did what I should. https://github.com/lou-bi/zmk-config/commit/4b95d82d875f61b568f7a87b878df674533644f1#annotation_16243079431

What are we supposed to do with this build.yaml you linked ?

And will this be fixed in a later released ? Im more comfortable using default build params. For no reasons propably.

ObiWanKeoni, 2023-12-08:
@lou-bi it looks like you've modified your `build.yaml` instead of your `.github/build.yml`, I think reverting your first change and applying Pete's fix will help move you along

> And will this be fixed in a later released ? Im more comfortable using default build params. For no reasons propably.

Yep, assuming that's what @petejohanson referred to when saying "If it works for you I'll get this into a PR."

caksoylar, 2023-12-08:
You can also try moving the nodefree folder into `config/` and refer to that path in the `#include`, I don't see a reason why that shouldn't work.

lou-bi, 2023-12-08:
@ObiWanKeoni , oh, that explain a lot of my previous trouble...
I can confirm too that @petejohanson 's fix did solve my issue !
Didn't try the @caksoylar one.