zmkfirmware/zmk #2063: glove80 realtek RTL8852AE bluetooth connectivity issues

saladbowlwithdressing, 2023-12-07:
hello there, new user here. whenever I try to connect my glove80 to windows there seem to be constant connectivity issues. whenever it does connect windows shows a flickering message between connected and paired in the bluetooth settings page. 

i applied the newest firmware and tried resetting the device multiple times but nothing seems to work.

also, to rule out if my keyboard was broken or not i tried connecting to my mac and it  connects without any issues every single time. i have yet to try it out on linux.

in the discord community for the glove80 multiple users with the realtek RTL8852AE chipset seem to have the same issue, so maybe its related to the chipset? 

thank you!

PRUDHVI69, 2023-12-20:
Can you give me some detailed information? i'm new here but i hope i can fix it 
Thank you :)

caksoylar, 2023-12-20:
This is probably covered by the tweaks and testing that is currently going on in ZMK and MoErgo Discords. Curren recommendation is to use try the `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` flag documented in https://zmk.dev/docs/config/bluetooth.

joeyv120, 2025-04-23:
I've experienced this intermittently with my nice!nano v2 split on both Windows 11 and Linux Mint 22. The only thing that seems to fix it is clearing the bluetooth profile from the keyboard & the host, and re-pairing. When this issue comes up I can still activate the profile that connects to my iOS device and successfully connect to it. My config is here: https://github.com/joeyv120/zmk-config-microdox