zmkfirmware/zmk #2096: Does ZMK support Hall Effect Sensor based Hardware?

SnappierSoap318, 2024-01-06:
can I use ZMK for an Analog Hall effect sensor-based keyboard using the Arduino ESP32-S3?

joelspadin, 2024-01-06:
There's no reason it couldn't be supported if you write a new key scan driver for it, but there is not currently any support for Hall effect keyboards. Also note that https://docs.zephyrproject.org/latest/boards/xtensa/esp32s3_devkitm/doc/index.html does not list the radio as a supported feature, so I'm not sure if Zephyr supports wireless on that controller yet.

(You could also ask on the Discord server. The only people watching the issues here are the code maintainers.)

SnappierSoap318, 2024-01-06:
Arduino Nano ESP32-S3 does have Wifi and BLE, so i might need to figure out a way to add that as well. ahh thanks!

closed 2024-01-06 by SnappierSoap318

myst729, 2024-06-03:
@SnappierSoap318 Any updates? Would like to try magnetic switches too.

gideontonkinson, 2024-07-24:
@joelspadin I'm interested contribuitng or forking the repo and creating the key scanner driver and I need to get familiar with the ZMK project, so do you have any hints on where to get started with that? If not I'll work through it and find it eventually. @myst729 if I get anywhere on it and especially if I finish it I'll come back here and let yah know.

joelspadin, 2024-07-24:
The existing key scan drivers are defined at https://github.com/zmkfirmware/zmk/tree/main/app/module/drivers/kscan. Reading up on [Zephyr drivers](https://docs.zephyrproject.org/latest/kernel/drivers/index.html) would also help.