zmkfirmware/zmk #2100: Reason for no support of 8bit platforms?

ArturGajowy, 2024-01-06:
@petejohanson states here https://github.com/zmkfirmware/zmk/issues/1110#issuecomment-1148532707:

> Zephyr doesn't support those 8-bit devices at all,not never will.

I was wondering about the reasons for this. Is that merely for historical reasons, or by design?
Is there anything in ZMK that would prevent porting it for, say, ATmega32U4?

If somebody tried - would that be a welcome contribution?
Would a fork providing that be considered friendly?

caksoylar, 2024-01-07:
ZMK is built on the Zephyr RTOS project (the Z in ZMK), which provides hardware architecture support. ZMK itself wouldn't add support for new architectures, that question should be aimed at Zephyr (https://github.com/zephyrproject-rtos/zephyr/) instead. My unqualified impression is that AVR chips just don't have enough resources to run Zephyr, even if you got it working with 8 bit ISA.

ArturGajowy, 2024-01-07:
Thank you, that's very helpful!

closed 2024-01-07 by ArturGajowy

ArturGajowy, 2024-01-07:
Having looked into Zephyr a bit (its scope, size, and supported architectures) I find my original question a bit funny :D