zmkfirmware/zmk #2101: ZMK macro will not build, even using the code from ZMK documentation

AkAlpenglow, 2024-01-06:
Hello,

I have a Kinesis Advantage 360 Pro, and have been able to successfully remap a few keys, but I'm struggling to make any macro that requires pressing two or more keys. I've tried updating the macros.dtsi file a few dozen different ways, and can't get the build to successfully complete, EVEN when directly copy-pasting the example code from the ZMK documentation on macros found here:
https://zmk.dev/docs/behaviors/macros

For example, when I try to use this snippet of the bindings list:

bindings
                = <&macro_press &kp LSHFT>
                , <&macro_tap &kp Z &kp M &kp K>
                , <&macro_release &kp LSHFT>
                ;

I get this error:
devicetree error: /__w/Adv360-Pro-ZMK-Germer/Adv360-Pro-ZMK-Germer/config/macros.dtsi:5 (column 17): parse error: expected number or parenthesized expression

My code currently looks like this:

macro_emoticonmenu: macro_emoticonmenu{
compatible = "zmk,behavior-macro";
label = "macro_emoticonmenu";
#binding-cells = <0>;
bindings
                = <&macro_press &kp LGUI &kp LS>
                , <&macro_tap &kp N4>
                , <&macro_release &kp LGUI &kp LS>
};
macro_screenshot: macro_screenshot{
compatible = "zmk,behavior-macro";
label = "macro_screenshot";
#binding-cells = <0>;
bindings
                = <&macro_press &kp LGUI &kp LC>
                , <&macro_tap &kp SPACE>
                , <&macro_release &kp LGUI &kp LC>
};

What I'm trying to do is have two macros, one to do a screenshot (press and hold LGUI and LS, then tap N4), and one to bring up the emoticon menu (press and hold LC and LGUI, then tap SPACE).

I have tried this with sticky keys, nesting sticky keys, etc. and nothing seems to work. Can anyone help?? I've even looked through over 100 forks off of the main repo to see how others have done their macros to find examples, and their code doesn't build, either.

joelspadin, 2024-01-06:
`LS` and `LC` are function macros that wrap another keycode, e.g. `LS(A)` for shift + A. They cannot be used on their own. You need to use the regular keycodes such as `LSHFT` and `LCTRL` instead.

If it still doesn't build when using the correct key codes, can you link to your config repo? It could also be something else entirely, such as using a fork or old version of ZMK that doesn't support macros.

AkAlpenglow, 2024-01-07:
Sure thing: here's the path to my repo:
https://github.com/AkAlpenglow/Adv360-Pro-ZMK-Germer/tree/V3.0

I updated the key codes and it still doesn't build. I'm forked off of ZMK V3.0, so it should work.. 

Thank you for helping me on this!

joelspadin, 2024-01-07:
The error is
> devicetree error: /__w/Adv360-Pro-ZMK-Germer/Adv360-Pro-ZMK-Germer/config/macros.dtsi:9 (column 1): parse error: expected ';' or ','

You're missing a semicolon at the end of line 8.

AkAlpenglow, 2024-01-07:
Thanks for this. Was looking too closely at it to see the obvious. Cheers!!

closed 2024-01-07 by AkAlpenglow