zmkfirmware/zmk #2118: grave not working

cw-20022326, 2024-01-16:
When you press the grave key on a Mac, § is entered. What should I do to solve this?

caksoylar, 2024-01-16:
Assuming you are using a keyboard locale on macOS which actually should have grave there, sounds like you have the wrong keyboard type selected, see https://support.apple.com/guide/mac-help/specify-a-keyboard-type-mchlp2886/14.0/mac/14.0

When setting up the keyboard in macOS, it might have asked you to press the key next to left shift to identify the type. ANSI keyboard would have Z there.

cw-20022326, 2024-01-16:
저는 ansi 키보드를 설정하였습니다. 키보드 유형 변경을 통해 ANSI로 다시 설정하였지만 동일한 §가 나타납니다.

caksoylar, 2024-01-16:
What is your keyboard layout in the OS?

cw-20022326, 2024-01-16:
It seems that the Bluetooth connection between the keyboard and mouse was incorrect. I disconnected the mouse and re-established it, and it output normally.
thank you

closed 2024-01-16 by cw-20022326