zmkfirmware/zmk #2125: XIAO BLE sleep current on zephyr 3.5.0

ldab, 2024-01-21:
> I am not sure if the power consumption is normal after waking up since I don't have a power profiler, but I'll track overall battery life.

I didn't flash the board you mentioned to my keyboard because the GPIOs are different, but I have build based on https://github.com/zmkfirmware/zmk/pull/1995 [my keyboard ](https://github.com/ldab/split-or-not-keyboard/blob/6d2f1f56749ac5582d0e0f6cebb14ece578a17ed/build.yaml#L21C12-L21C23) and the result is:

[image]

So using **https://github.com/zmkfirmware/zmk/pull/1995** the sleep current is 20uA vs ~1uA I measured not, but it seems to fix the sleep issue after second wake. I guess we can open an issue to track it and investigate more when 3.5.0 is merged, what do you think @caksoylar ?

_Originally posted by @ldab in https://github.com/zmkfirmware/zmk/issues/2091#issuecomment-1902239239_

carrefinho, 2024-04-20:
Can confirm my XIAO nRF52840 on 3.5 also sleeps at ~20µA

[image]

caksoylar, 2024-04-21:
While I don't have a PPK, I can confirm that my battery life has massively improved since 3.5 upgrade. My peripheral half (of a split) is especially better, where the battery life is more than 5x as long as before. That would be expected if the standby/sleep power usage decreased significantly.

closed 2025-03-31 by nmunnich