zmkfirmware/zmk #2132: OLEDs do not fit into the nRF52833 flash anymore

krikun98, 2024-01-25:
Ever since `b021d192` OLED builds do not fit into `nrfmicro_13_52833`:
```
west update && west build -p -b nrfmicro_13_52833 -d build/jorne_test/left -- -DSHIELD=jorne_left
...
/home/screamingwhale/.local/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `datas' will not fit in region `FLASH'
/home/screamingwhale/.local/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 4388 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
Full logs are available here: https://github.com/krikun98/zmk-config/actions/runs/7005103094/job/19054193000
Builds with `CONFIG_ZMK_DISPLAY=y` for the standard SSD1306 fail to fit into the flash.
If I revert that change:
```
diff --git a/app/src/display/Kconfig b/app/src/display/Kconfig
index a2029481..d419300b 100644
--- a/app/src/display/Kconfig
+++ b/app/src/display/Kconfig
@@ -41,7 +41,6 @@ choice ZMK_DISPLAY_STATUS_SCREEN
 config ZMK_DISPLAY_STATUS_SCREEN_BUILT_IN
     bool "Built in status screen"
     select LV_OBJ_LABEL
-    imply LV_USE_THEME_MONO
     imply ZMK_WIDGET_LAYER_STATUS
     imply ZMK_WIDGET_BATTERY_STATUS
     imply ZMK_WIDGET_OUTPUT_STATUS
```
The firmware builds fine on the current `main` branch.

joelspadin, 2024-01-25:
`imply LV_USE_THEME_MONO` means that the config isn't forced. If you set `CONFIG_LV_USE_THEME_MONO=n` in your .conf file, does that build again?

krikun98, 2024-01-25:
@joelspadin, it does, yes.

joelspadin, 2024-01-26:
Is that a good enough solution for you, or are you expecting it to work automatically? I'm not familiar enough with kconfig to know if this is something that can be set automatically in nrfmicro's Kconfig.defconfig.

Devices with constrained memory and flash space are probably always going to require some manual tweaking to get extra features to fit, but maybe there are other places we can reduce ROM use as well.

krikun98, 2024-01-26:
This is an acceptable workaround, but it didn't require any tweaking before the commit specified in the patch, and I doubt this was intentional.
@petejohanson mentioned on Discord that an option which would reduce memory usage is not getting applied early enough:
`Seems related to the theme, but also the `LV_CONF_MINIMAL` not actually applying early enough, so a *ton* of extra ZMK features got turned on too, but maybe that's because I enabled via menuconfig.`