zmkfirmware/zmk #2142: Backlight RGB config not compiling using Github Actions

Sabin-Adhikari, 2024-01-28:
Hi guys,
I am trying  to add RGB support to my ZMK firmware.
My underglow lights work fine, but I am having issues with the individual RGB setup.
I get the following errors during compilation.

`
FAILED: CMakeFiles/app.dir/src/backlight.c.obj 
ccache /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/tmp/tmp.6pDixcMQqQ/zmk/app/include -I/tmp/tmp.6pDixcMQqQ/zephyr/include -I/tmp/tmp.4etB087BIp/zephyr/include/generated -I/tmp/tmp.6pDixcMQqQ/zephyr/soc/arm/nordic_nrf/nrf52 -I/tmp/tmp.6pDixcMQqQ/zephyr/soc/arm/nordic_nrf/common/. -I/tmp/tmp.6pDixcMQqQ/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -I/tmp/tmp.6pDixcMQqQ/zephyr/subsys/bluetooth -I/tmp/tmp.6pDixcMQqQ/zephyr/subsys/usb/device -I/tmp/tmp.6pDixcMQqQ/zephyr/subsys/settings/include -I/tmp/tmp.6pDixcMQqQ/modules/hal/cmsis/CMSIS/Core/Include -I/tmp/tmp.6pDixcMQqQ/modules/hal/nordic/nrfx -I/tmp/tmp.6pDixcMQqQ/modules/hal/nordic/nrfx/drivers/include -I/tmp/tmp.6pDixcMQqQ/modules/hal/nordic/nrfx/mdk -I/tmp/tmp.6pDixcMQqQ/zephyr/modules/hal_nordic/nrfx/. -I/tmp/tmp.6pDixcMQqQ/modules/crypto/tinycrypt/lib/include -I/tmp/tmp.6pDixcMQqQ/zmk/app/module/include -I/tmp/tmp.6pDixcMQqQ/zmk/app/module/drivers/sensor/battery/. -isystem /tmp/tmp.6pDixcMQqQ/zephyr/lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -fno-strict-aliasing -Os -imacros /tmp/tmp.4etB087BIp/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /tmp/tmp.6pDixcMQqQ/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/tmp/tmp.6pDixcMQqQ/zmk/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/tmp/tmp.6pDixcMQqQ/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/tmp/tmp.6pDixcMQqQ=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -Wfatal-errors -MD -MT CMakeFiles/app.dir/src/backlight.c.obj -MF CMakeFiles/app.dir/src/backlight.c.obj.d -o CMakeFiles/app.dir/src/backlight.c.obj -c /tmp/tmp.6pDixcMQqQ/zmk/app/src/backlight.c
In file included from /tmp/tmp.6pDixcMQqQ/zephyr/include/zephyr/toolchain.h:50,
                 from /tmp/tmp.6pDixcMQqQ/zephyr/include/zephyr/devicetree/zephyr.h:15,
                 from /tmp/tmp.6pDixcMQqQ/zephyr/include/zephyr/devicetree.h:3755,
                 from /tmp/tmp.6pDixcMQqQ/zephyr/include/zephyr/device.h:38,
                 from /tmp/tmp.6pDixcMQqQ/zmk/app/src/backlight.c:7:
/tmp/tmp.6pDixcMQqQ/zephyr/include/zephyr/toolchain/gcc.h:77:36: error: static assertion failed: "CONFIG_ZMK_BACKLIGHT is enabled but no zmk,backlight chosen node found"
   77 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert(EXPR, "" MSG)
      |                                    ^~~~~~~~~~~~~~
/tmp/tmp.6pDixcMQqQ/zmk/app/src/backlight.c:25:1: note: in expansion of macro 'BUILD_ASSERT'
   25 | BUILD_ASSERT(DT_HAS_CHOSEN(zmk_backlight),
      | ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
[35/300] Building C object CMakeFiles/app.dir/src/workqueue.c.obj
[36/300] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/crc16_sw.c.obj
[37/300] Building C object CMakeFiles/app.dir/src/split/bluetooth/peripheral.c.obj
[38/300] Building C object CMakeFiles/app.dir/src/main.c.obj
[39/300] Building C object CMakeFiles/app.dir/src/rgb_underglow.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /tmp/tmp.4etB087BIp
Error: Process completed with exit code 1.
0s
`

My corne.conf file has following RGB config:
`
CONFIG_ZMK_RGB_UNDERGLOW=y
CONFIG_WS2812_STRIP=y
CONFIG_ZMK_BACKLIGHT=y
CONFIG_ZMK_BACKLIGHT_AUTO_OFF_IDLE=y
`

I am new to ZMK, so any help is appreciated. 
You can see the entire zmk config codebase in my repo - **zmk-conf-corne-v3**
Thanks in advance!

Sabin-Adhikari, 2024-01-29:
Okay. I will try removing it.

Sabin-Adhikari, 2024-01-29:
I just tested it. It worked on the master but not on the slave.

caksoylar, 2024-01-30:
Did you flash the right side with new firmware as well? If you did, it might be a hardware issue.

Sabin-Adhikari, 2024-01-31:
Yes I flashed both of the sides. I can see the master side is fully functioning with all lights, and the right side has only 6 lights glowing. Rest of the lights are not working. 
I have a multi meter but I am not sure how to check the lights. I could not find any resources online either

Sabin-Adhikari, 2024-01-31:
Hey guys, Sorry for your help. I figured the issue was because of weak soldering on one of the RGB light joints. I resoldered it and its now working as expected. Thanks loads!

closed 2024-01-31 by Sabin-Adhikari