zmkfirmware/zmk #2156: Unable to enter bootloader

BoweFlex, 2024-02-09:
I recently built a ferris sweep using nice!nanos with ZMK, and when I set up my keymap I neglected to provide a bootloader key (I knew what it was for but was nervous about accidentally pressing it and didn't realize how important it would be). I've made some changes to my keymap and built the .uf2 files, but I can't get either half of my keyboard to enter the bootloader and allow me to reflash it. I've tried doubletapping the reset button as well as double tapping the reset and ground pins on the nice!nanos, and it seems to soft reset as it will stop working and then reconnect but does not enter the bootloader. Are there any other options for me to resolve this? Even if I need to factory reset the MCUs and start from scratch

Luerl21, 2024-02-09:
Maybe you need to press faster

BoweFlex, 2024-02-09:
I hate to admit it, but I think that was the problem.. After a few more attempts hitting the pins faster, I was able to reset both halves. Thank you!

closed 2024-02-09 by BoweFlex