zmkfirmware/zmk #2160: Support AlienFX over USB

feature-engineer, 2024-02-11:
Dell has a USB protocol which is used by some games to modify the underglow (under the keycaps) such that only keys used by the game are lit up.

Reverse engineering of this protocol was done by T-Troll (https://github.com/T-Troll/alienfx-tools/tree/master/AlienFX-SDK), and can be used to add support for this feature. 

It could even be extended to wireless mode, if a service + virtual USB keyboard driver are used to monitor for this USB protocol and send it over websockets.

joelspadin, 2024-02-11:
I don't think the lighting code in ZMK is really fleshed out well enough to support that type of thing yet, but is there a specific reason you would want to use Dell's proprietary protocol specifically? Personally, I'd rather try to find an open protocol to use (or create a new one and integrate it with https://openrgb.org/)

feature-engineer, 2024-02-12:
@joelspadin The proprietary protocol is already utilized by games (and is also used by Roccat, not just Alienware).
While an open standard would be great, it would only work if games were to use it. Maybe if Valve or System76 got behind such a standard it could work - with a free implementation and a dedicated dev who would add it free of charge to games in development, like nvidia does with new features they develop.

But as it currently stands, I think our best bet of having our lights sync with games is using what's currently used in games.

feature-engineer, 2024-02-12:
Hmm... On second thought, we could just create a lightfx.dll with the same function entries, and make it implement whatever protocol we would like. Unless I'm missing something.

There's https://github.com/Archomeda/lightfx-extender/tree/master 
Which is a discontinued project, but it has the dll signature needed. 

Using this, it would be easy to support any protocol and make it work with AlienFX.

So I guess it would be enough to implement and open standard protocol or invent a new one if one doesn't exist, and support AlienFX on the dll level.