zmkfirmware/zmk #2163: Asymmetric split keyboard

berikiushi, 2024-02-13:
Is it possible to create firmware for an asymmetric split keyboard whose halves have different wiring boards and use different controller pins for columns and rows?

left: 29 keys + roll encoder + nice!nano + nice!view
right: 43 keys + roll encoder + nice!nano

[image]

[image]

petejohanson, 2024-02-13:
Of course. In this case the only thing you'd really have in the common .dtsi would be the matrix transform.

berikiushi, 2024-02-13:
> Of course. In this case the only thing you'd really have in the common .dtsi would be the matrix transform.

Thanks for the reply, it's encouraging.
Just now I have found several examples of asymmetric keyboards in the repository, I will study them in more detail.

closed 2024-02-13 by berikiushi