zmkfirmware/zmk #2169: SplitKB Aurora Helix RGB pin interference

sidju, 2024-02-19:
# Background and problem:
The Aurora Helix uses pull-up/pull-down on the RGB pin as a side indicator for QMK. When I used latest ZMK to build firmware with the RGB enabled ([config](https://github.com/sidju/zmk-helix-config)) the left side seemed to have some interference on the RGB pin, which I eventually narrowed down to be caused by the pull-up.

# Suspected cause:
I couldn't find any declaration of pin configuration in the Aurora Helix shield so I couldn't confirm this, but it is my suspicion that the uC pin is set to pull-down in ZMK and that causes the interference due to it being kinda floating.

# Other potential cause:
If you can disprove this ZMK pin config theory it is probably the nice!nano that doesn't handle overriding a pull-up with that resistance sufficiently to not be affected by the VCC ripple, in which case I'll get in touch with SplitKB to change the board design or build instructions.

sidju, 2024-02-19:
I have already cut the trace connecting the VCC resistor to the RGB pin and that solved the issue for me. The pull-down resistor didn't cause any issues on the other half, so I've left that as is for now.

I'm just looking into how this can best be fixed; if a software fix could be applied that allows other to follow the build-guide as it is now, or if the build guide needs to have a "If you aren't using a nice!nano or ZMK" clause on soldering those resistors.

I have already contacted SplitKB and linked them this issue, so they can see the conclusion on this end.

thomasbaart, 2024-02-29:
For reference: The RGB pin serves a dual purpose in QMK, where it's used for side detection first, and then switched over to RGB later during the initialisation process.

As side detection for ZMK is done differently, the pullup can be omitted. I'll be sure to add a note to the build guide for this.

sidju, 2024-02-29:
Great. We can consider this to have been a HW issue with a HW solution in the guide. Feel free to close this issue. (I won't, as many repos prefer to manage their own issues.)

closed 2024-02-29 by caksoylar