zmkfirmware/zmk #2170: Mention "CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y" in "Bluetooth#Troubleshooting"

dennisorlando, 2024-02-19:
Hi, I just fixed a bluetooth connectivity issue by following instructions in #1487 
Updating the documentation with the fixes described in the issue might be worth it, although I don't actually know what `CONFIG_ZMK_BLE_EXPERIMENTAL_CONN=y` does and the reason why I can't connect to my split keyboard without it.

slabua, 2024-02-24:
Wait for it. A lot of people who have done the same in that thread are again encountering the same connection issues.

caksoylar, 2024-02-27:
My impression is that Zephyr 3.5 upgrade helped a lot with issues, but this setting (which only disables PHY 2M at the moment) might still be necessary for some people (and it no longer seems to hurt with some Macs). We need more data points after the upgrade, then we can see if we should recommend it in troubleshooting or perhaps default to it.

caksoylar, 2024-07-25:
Fixed by #2387.

closed 2024-07-25 by caksoylar

slabua, 2024-07-26:
Updating the documentation does not fix the problem, though.

caksoylar, 2024-07-26:
> Updating the documentation does not fix the problem, though.

This Github issue is specifically about missing documentation about the Kconfig symbol, so not sure what the complaint is about.