zmkfirmware/zmk #2173: Proof of Concept: ZMK works with ESP32-S3

tywtyw2002, 2024-02-24:
This week, I did some tests on my ESP32-S3 (ESP32-S3R2) corne/corneish clone with the ZMK firmware.

- 16M SPI FLASH
- 2M Internal PRAM

The positive outcome is that the ESP32-S3 SoC can run ZMK directly without any code modifications with Zephyr v3.6.

- [X] Single Keyboard Bluetooth
- [X] Split Keyboard Bluetooth

At present, the ESP32-S3 can only communicate with the host via Bluetooth. This is because Zephyr v3.6 does not yet support ESP32 USB OTG.

A few things need to be mentioned.
- The ESP32-S3 does not yet support `usbd`; instead, it uses `usb_serial` for jtag/serial. To enable `ZMK_USB_LOGGING`, you need to comment out `USB_DEVICE_STACK` and `USB_CDC_ACM` in `Kconfig`.
- Regarding `Settings`, only the `FCB` is functional. The `NVS` leads to a core dump. I haven't thoroughly investigated this issue, but my best guess is that there are some bugs related to the `NVS` API at the `hal_espressif` level.
- It's possible to use UF2 firmware on the esp32-s3 with the `tinyuf2` bootloader. However, I was unable to successfully build the UF2 ZMK firmware.

I would like to note that I don't believe the ESP32-S3 is a suitable MCU for keyboards at this point. This is primarily because Zephyr does not fully support the ESP32 yet, particularly the USB OTG. I'm also unsure about the practicality of a keyboard that connects via BLE but is powered by USB.

Regarding power consumption, my keyboard uses approximately 0.1A@5V when active.

The ESP32-C3 should also works with ZMK as it has ble5

Reference: https://github.com/tywtyw2002/zmk/tree/z/esp32

petejohanson, 2024-02-24:
Thanks for testing! Does sound like a few things need to mature just a tad more for it to be considered a real viable option though, as you noted.

septatrix, 2024-08-20:
Support for USB OTG is planned though no specific timeline yet. See https://github.com/zephyrproject-rtos/zephyr/issues/29394

telok-ry, 2024-11-21:
Any updates on this? I am currently considering a portable kb (80 keys) and S3 would be great since it has enough IO for this.. No need for USB OTG but as there is no branch I am a bit uncomfortable on committing on S3

deffener, 2024-11-21:
Any updates? I am considering a design of a portable kb with 80 keys, S3 would have a nice amount of IO.. Also no need for OTG since bt works.. However there is not branches so I am a bit afraid of committing to S3.. (posted the same comment on wrong account previously :))

deffener, 2024-11-24:
Probably also abandoning S3 for zmk as I have some nrf52 also and remembered that I also have some io-expanders so I don't need the IO from the mcu anyway ^^

Thanks for the info anyway!

klesh, 2025-02-12:
@tywtyw2002 Care to share the reason you abandoned the chip? Was it the power consumption or USB not working?

tywtyw2002, 2025-02-12:
> [[image]@tywtyw2002](https://github.com/tywtyw2002?rgh-link-date=2025-02-12T14%3A05%3A54.000Z) Care to share the reason you abandoned the chip? Was it the power consumption or USB not working?

Both of them and I do not want to spend too much time on it.

compared with NRF52, the ESP3-S3 need spend 10x more time.

The reason that I choose the ESP3-S3 because the NRF52 hard to solder.

klesh, 2025-02-13:
@tywtyw2002 Thank you for sharing your insights. I'll probably also stick with the NRF52.

RamKromberg, 2025-02-22:
> I would like to note that I don't believe the ESP32-S3 is a suitable MCU for keyboards at this point. This is primarily because Zephyr does not fully support the ESP32 yet, particularly the USB OTG. I'm also unsure about the practicality of a keyboard that connects via BLE but is powered by USB.

@tywtyw2002 There's some work being done on this: https://github.com/zephyrproject-rtos/zephyr/issues/29394#issuecomment-2634634350

tywtyw2002, 2025-02-26:
no longer work on it. closed.

closed 2025-02-26 by tywtyw2002

efogdev, 2025-03-03:
Sorry for (kinda) offtopic but I thought that NRF SoCs don't support USB OTG?

Nick-Munnich, 2025-03-03:
> Sorry for (kinda) offtopic but I thought that NRF SoCs don't support USB OTG?

It doesn't, but it does support being used as a USB peripheral, for interrupts in particular. The ability to be a USB peripheral with interrupts is what we mainly need for a chip to be used as a USB keyboard/mouse.