zmkfirmware/zmk #2174: nice nano v2 + miryoku: key sticks, crashes, requires power to reset

willtalmadge, 2024-02-25:
I just setup a typeractive corne with nice nano v2. I forked the miryoku zmk repo and built firmware with their github actions and installed it to the keyboards. They are paired to macos via BLE. Every 10-20 minutes, as I'm typing a key will stick (for example it'll type out xxxxxxxxxxxxxxxxxxxxx) then it stops typing and the keyboard is unresponsive. Power cycling both halves has no effect. If I plug one half into usb-c power and then unplug it after a few seconds, it seems to reset the state of the keyboard and it works as normal for another 10 or so minutes.

Any thoughts what might be going on? How do I get logs to troubleshoot this? I'm new to zmk keyboards, hoping to get this fixed so I can use this for more than typing practice. Thanks!

Firmware fork https://github.com/willtalmadge/miryoku_zmk/tree/master

caksoylar, 2024-02-25:
Are you the battery isn't an issue? "Plug into USB, plug off, keeps working for 10 minutes" sure sounds like the battery gets some charge and dies again.

willtalmadge, 2024-03-24:
I think that was it, thanks for the tip. Seems like one of my boards is draining a lot faster than I expected.

closed 2024-03-24 by willtalmadge